ros2/kilted_tutorial_party #327: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-05:
Hi, I worked on this ticket

I built bondcpp from source (https://github.com/ros/bond_core) which depends on rclcpp and rclcpp_lifecycle.

It works. Full log output below:

```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.14.0-pre.5.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************

C:\Program Files\Microsoft Visual Studio\2022\Preview>call "\Program Files\rti_connext_dds-7.3.0\resource\scripts\rtisetenv_x64Win64VS2017.bat"
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.

                           RTI Connext DDS 7.3.0

###############################################################################

C:\Program Files\Microsoft Visual Studio\2022\Preview>set CONNEXTDDS_DIR=C:\Program Files\rti_connext_dds-7.3.0

C:\Program Files\Microsoft Visual Studio\2022\Preview>set RTI_LICENSE_FILE="C:\Users\suche\Downloads\rti_license.dat"

C:\Program Files\Microsoft Visual Studio\2022\Preview>set RMW_IMPLEMENTATION=rmw_connextdds

C:\Program Files\Microsoft Visual Studio\2022\Preview>ros2 doctor --report | grep middleware
'ros2' is not recognized as an internal or external command,
operable program or batch file.

C:\Program Files\Microsoft Visual Studio\2022\Preview>cd C:\pixi_ws

C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_connextdds

(pixi_ros2_rolling) C:\pixi_ws>cd \ros2_ws

(pixi_ros2_rolling) C:\ros2_ws>colcon build --merge-install
Starting >>> bond
Starting >>> smclib
Finished <<< smclib [8.69s]
Finished <<< bond [28.7s]
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [5.48s]
--- stderr: bondcpp
ERRORFailed to load RTI Connext DDS Micro
---
Finished <<< bondcpp [21.7s]
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [6.09s]
[Processing: test_bond]
--- stderr: test_bond
ERRORFailed to load RTI Connext DDS Micro
---
Finished <<< test_bond [37.5s]

Summary: 6 packages finished [1min 28s]
  2 packages had stderr output: bondcpp test_bond

(pixi_ros2_rolling) C:\ros2_ws>colcon test --merge-install
Starting >>> bond
Starting >>> smclib
Finished <<< smclib [7.48s]
Finished <<< bond [17.7s]
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondcpp [3.81s]
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.42s]
Finished <<< test_bond [15.2s]
--- stderr: bondpy

============================== warnings summary ===============================
test/test_flake8.py::test_flake8
  Warning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< bondpy [25.1s]

Summary: 6 packages finished [43.2s]
  1 package had stderr output: bondpy

(pixi_ros2_rolling) C:\ros2_ws>colcon test-result
Summary: 262 tests, 0 errors, 0 failures, 60 skipped

(pixi_ros2_rolling) C:\ros2_ws>
```

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde