ros2/kilted_tutorial_party #328: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Barry-Xu-2018, 2025-05-08:
PLATFORM INFORMATION
Hardware: Raspberry Pi 4 Model B
System : Ubuntu 24.04.2
processor : aarch64

RMW MIDDLEWARE
middleware name : rmw_connextdds

ROS 2 INFORMATION
distribution name : kilted
distribution type : ros2
distribution status : pre-release
release platforms : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

#### Test result
Test passed

#### Test Steps
Prepare ROS 2 Kilted environment (Refer to https://docs.ros.org/en/kilted/Installation/Ubuntu-Install-Debs.html)

Install the additional required packages
```bash
$ sudo apt install -y rti-connext-dds-7.3.0-ros ros-kilted-rmw-connextdds
```

Use `https://github.com/ros/bond_core` to do test since it depends on rclcpp and rclcpp_lifecycle.  
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ mkdir -p kilted_ws/src && cd kilted_ws/src
$ git clone https://github.com/ros/bond_core.git
$ cd ..
$ colcon build --symlink-install
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [8.57s]                                                  
[Processing: bond]                              
Finished <<< bond [40.6s]                    
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [7.92s]                                                  
[Processing: bondcpp]                                  
Finished <<< bondcpp [1min 3s]                              
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [5.43s]                                                             
[Processing: test_bond]                                  
[Processing: test_bond]                                      
Finished <<< test_bond [1min 29s]                              

Summary: 6 packages finished [3min 14s]

$ colcon test
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [9.92s]                                     
Finished <<< bond [38.0s]            
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondcpp [13.7s]                                         
Starting >>> bond_core
Starting >>> test_bond                                                
Finished <<< bond_core [0.12s]                                                           
--- stderr: bondpy                                                         

=============================== warnings summary ===============================
test/test_flake8.py::test_flake8
test/test_flake8.py::test_flake8
test/test_flake8.py::test_flake8
test/test_flake8.py::test_flake8
  Warning: This process (pid=38751) is multi-threaded, use of fork() may lead to deadlocks in the child.

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< bondpy [27.7s]
Finished <<< test_bond [20.4s]                 

Summary: 6 packages finished [1min 13s]
  1 package had stderr output: bondpy

$ colcon test-result
Summary: 262 tests, 0 errors, 0 failures, 60 skipped
```

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde