ros2/kilted_tutorial_party #329: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-06:
I understand there's been some issues with Connext in the debians (for both amd64 and arm64), but that they were recently resolved, so I can pick this up.

danielcranston, 2025-05-06:
Ah, this ticket is largely RMW agnostic 😅 Anyways,

Some small comments on https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-RTI-Connext-DDS.html:

> Connext DDS is included when installing `rmw_connextdds` using apt

A bit of a nit, but to me this is implying that an apt package with that name exists, but afaict it does not. I installed `rti-connext-dds-7.3.0-ros` and `ros-kilted-rmw-connextdds` as instructed a bit further down.

> ROS 2 users can install a non-commercial-use version of the RTI Connext DDS libraries for x86_64 Linux from the ROS apt repository [...]

This one has been mentioned elsewhere, but now `arm64` is supported as well so the `x86_64` comment can be removed.

- [x] **Local workspace overlay works**

```bash
$ ros2 pkg create --build-type ament_cmake test_connext_pkg --node-name talker --dependencies rclcpp geometry_msgs
$ ros2 pkg create --build-type ament_python test_connext_pkg_py --node-name listener --dependencies rclpy geometry_msgs
$ cd .. # ws root
$ colcon build --packages-select test_connext_pkg test_connext_pkg_py
$ colcon test --packages-select test_connext_pkg test_connext_pkg_py
$ colcon test-result
Summary: 12 tests, 0 errors, 0 failures, 2 skipped
```

---

> TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

To test this I made the above C++ and Python nodes actually publish and subscribe to a geometry_msgs/Pose (replicating the talker/listener demo). Rebuild and works fine

```bash
# Terminal 1
$ source /opt/ros/kilted/setup.bash 
$ source ~/colcon_ws/install/setup.bash 
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 run test_connext_pkg talker 
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746512140.969606453] [talker]: Publishing
[INFO] [1746512141.469664455] [talker]: Publishing
[INFO] [1746512141.969474147] [talker]: Publishing
...
```

```bash
# Terminal 2
$ source /opt/ros/kilted/setup.bash 
$ source ~/colcon_ws/install/setup.bash 
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 run test_connext_pkg_py listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746512155.497218775] [pose_listener_node]: I heard: geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=1.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=1.0, y=0.0, z=0.0, w=1.0))
...
```

danielcranston, 2025-05-12:
I think this can be closed. The first comment is just a wording nit, the second was addressed by https://github.com/ros2/ros2_documentation/pull/5574.

closed 2025-05-12 by ahcorde