ros2/kilted_tutorial_party #330: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-05:
Does this have to be a ROS 2 Kilted package, or can it be my "Humble-Dave" robot package?

(Also did overlay of demo_nodes_py - see below my robot test)

I ported my ROS 2 Humble Dave robot to be "ROS 2 Kilted Dave" and it functioned correctly - reading battery via i2c sensor, getting off the dock when fully charged and returning to the dock when battery level is low.

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ cmds/start_battery.sh 

*** Switching to ~/KiltedDave/ros2ws

*** Sourcing /opt/ros/kilted/setup.bash

*** Sourcing install/setup.bash

*** Starting Battery Node
*** ros2 run dave battery_node &
ubuntu@kilteddave:~/KiltedDave/ros2ws$ RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.

ubuntu@kilteddave:~/KiltedDave/ros2ws$ cmds/echo_battery_state.sh 

*** Switching to ~/KiltedDave/ros2ws

*** Sourcing /opt/ros/kilted/setup.bash

*** Sourcing install/setup.bash

*** Capturing one battery state topic
*** ros2 topic echo --once /battery_state
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
header:
  stamp:
    sec: 1746449857
    nanosec: 927043671
  frame_id: base_link
volts: 12.201589584350586                                                 <<<<---- my robot's battery_node.py correctly running under kilted
milliamps: 37.317073822021484
watts: 0.19430893659591675
watthours: 0.0009174571023322642
charging: false
capacity: 24
percent: 100
last_charge: 0.0
last_discharge: 0.0
---
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ls src/
apriltag_msgs  dave             explorer_wanderer  gpg3_navigation2  ros2_gopigo3_msg   ros2_libcamera_pub  teleop_gopigo3_keyboard
apriltag_ros   dave_interfaces  gpg3_cartographer  nav2_gopigo3      ros2_gopigo3_node  scan_client         ydlidar_ros2_driver
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ls src/dave
dave  package.xml  resource  setup.cfg  setup.py  test
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ls src/dave/dave
battery_node.py  dave_node.py  docking_node.py  drive_node.py  __init__.py  say_node.py  say_svc_client.py  test_dock.py
ubuntu@kilteddave:~/KiltedDave/ros2ws$ more src/dave/dave/battery_node.py 
#!/usr/bin/env python3

# FILE: battery_node.py

"""

  This node uses the EasyINA219 sensor driver 
  to monitor the GoPiGo3 Li-Ion battery and 
  publish /battery_status topic with
  - volts
  - milliamps
  - watts
  - watthours  (since charge/discharge state change)
[... 152 lines cut ...]

Summary: 1 package finished [6.93s]
ubuntu@kilteddave:~/KiltedDave/ros2ws$ . ss.sh 

*** Switching to ~/KiltedDave/ros2ws

*** Sourcing /opt/ros/kilted/setup.bash

*** Sourcing install/setup.bash
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 run demo_nodes_py listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746450920.329450219] [listener]: [overlay listener] I heard: [Hello World: 0]
[INFO] [1746450921.223096027] [listener]: [overlay listener] I heard: [Hello World: 1]
[INFO] [1746450922.223162869] [listener]: [overlay listener] I heard: [Hello World: 2]
[INFO] [1746450923.225919973] [listener]: [overlay listener] I heard: [Hello World: 3]
[INFO] [1746450924.224568812] [listener]: [overlay listener] I heard: [Hello World: 4]
[INFO] [1746450925.225377842] [listener]: [overlay listener] I heard: [Hello World: 5]
[INFO] [1746450926.225382992] [listener]: [overlay listener] I heard: [Hello World: 6]
[INFO] [1746450927.224806001] [listener]: [overlay listener] I heard: [Hello World: 7]
^C[ros2run]: Received signal:  Interrupt
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

slowrunner, 2025-05-06:
Forgot to include:

## Checks
* [x]  **Local workspace overlay works**

ahcorde, 2025-05-06:
Thank you for testing

closed 2025-05-06 by ahcorde