ros2/kilted_tutorial_party #333: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-04:
Hello, I built bondcpp (https://github.com/ros/bond_core) from source. I followed https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html#id7

However, I faced a few issues. I got errors similar to the ones mentioned in https://robotics.stackexchange.com/questions/90768/ros2-windows10-colcon-build-fails-on-ros2-example-ws-tutorial

After following, I got more errors which were the same as the ones mentioned in https://robotics.stackexchange.com/questions/93446/could-not-find-any-instance-of-visual-studio

Outside of the documentaiton, I had to install visual studio with then had to make sure the `desktop development with C++` from `tools and features` were installed for the build to finally work.

After doing the above, I built the workspace and the tests passed successfully.

Entire log output is below:

```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.14.0-pre.5.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'

C:\Program Files\Microsoft Visual Studio\2022\Preview>cd C:\pixi_ws

C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

(pixi_ros2_rolling) C:\pixi_ws>set RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_cyclonedds_cpp

(pixi_ros2_rolling) C:\pixi_ws>cd \ros2_ws

(pixi_ros2_rolling) C:\ros2_ws>colcon build --merge-install
Starting >>> bond
Starting >>> smclib
Finished <<< smclib [7.23s]
Finished <<< bond [22.7s]
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [3.52s]
Finished <<< bondcpp [16.9s]
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [5.50s]
[Processing: test_bond]
Finished <<< test_bond [35.5s]

Summary: 6 packages finished [1min 15s]

(pixi_ros2_rolling) C:\ros2_ws>colcon test --merge-install
Starting >>> bond
Starting >>> smclib
Finished <<< smclib [9.08s]
Finished <<< bond [18.0s]
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondcpp [3.88s]
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.42s]
Finished <<< test_bond [14.6s]
--- stderr: bondpy

============================== warnings summary ===============================
test/test_flake8.py::test_flake8
  Warning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< bondpy [24.5s]

Summary: 6 packages finished [42.9s]
  1 package had stderr output: bondpy

(pixi_ros2_rolling) C:\ros2_ws>colcon test-result
Summary: 262 tests, 0 errors, 0 failures, 60 skipped

(pixi_ros2_rolling) C:\ros2_ws>
```

suchetanrs, 2025-05-04:
https://github.com/ros2/kilted_tutorial_party/issues/346#issuecomment-2849142475