ros2/kilted_tutorial_party #334: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

boby2965, 2025-05-04:
Hi, I worked on this ticket

# My Setup

```
ubuntu@kilted-desktop:~/cyclone_bond_ws$ dpkg --print-architecture
arm64

ubuntu@kilted-desktop:~/cyclone_bond_ws$ lsb_release -d
No LSB modules are available.
Description:	Ubuntu 24.04.2 LTS
 
ubuntu@kilted-desktop:~/cyclone_bond_ws$ sudo echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp

```

I built bondcpp from source (https://github.com/ros/bond_core) which depends on rclcpp and rclcpp_lifecycle.
when installing there was a warning but the build completed successfully.
```
ubuntu@kilted-desktop:~/cyclone_bond_ws$ colcon build --symlink-install --packages-up-to bondcpp
Starting >>> bond    
Starting >>> smclib                
Finished <<< smclib [13.9s]                                                     
[Processing: bond]                             
Finished <<< bond [50.8s]                       
Starting >>> bondcpp
[Processing: bondcpp]                                  
--- stderr: bondcpp                                     
In file included from /home/ubuntu/cyclone_bond_ws/src/bond_core/bondcpp/src/bond.cpp:31:
/home/ubuntu/cyclone_bond_ws/src/bond_core/bondcpp/include/bondcpp/bond.hpp:270:8: warning: private field 'deadpublishing_timer_reset_flag_' is not used [-Wunused-private-field]
  270 |   bool deadpublishing_timer_reset_flag_ {false};
      |        ^
1 warning generated.
---
Finished <<< bondcpp [54.0s]
                           
Summary: 3 packages finished [1min 45s]
  1 package had stderr output: bondcpp

```

All the tests passed as well

```
ubuntu@kilted-desktop:~/cyclone_bond_ws$ colcon test --packages-up-to bondcpp
colcon test-result --verbose
Starting >>> bond    
Starting >>> smclib                
Finished <<< smclib [10.6s]                                       
Finished <<< bond [38.5s]            
Starting >>> bondcpp
Finished <<< bondcpp [7.19s]           

Summary: 3 packages finished [46.4s]
Summary: 242 tests, 0 errors, 0 failures, 59 skipped
```

boby2965, 2025-05-04:
i have also created a dummy package and run executables 

```
ubuntu@kilted-desktop:~/cyclone_bond_ws$ colcon build --packages-select bondcpp_dummy_user
Starting >>> bondcpp_dummy_user
Finished <<< bondcpp_dummy_user [22.9s]                       

Summary: 1 package finished [23.4s]
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde