ros2/kilted_tutorial_party #340: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-02:
- [X] **Local workspace overlay works**

```
$ mkdir -p colcon_ws/src
$ cd colcon_ws/src
$ ros2 pkg create --build-type ament_cmake kilted_tutorial_party_340 --node-name my_node
$ cd ..
$ source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash 
$ colcon build
Starting >>> kilted_tutorial_party_340
Finished <<< kilted_tutorial_party_340 [0.57s]                     

Summary: 1 package finished [0.87s]

$ colcon test
Starting >>> kilted_tutorial_party_340
Finished <<< kilted_tutorial_party_340 [1.22s]          

Summary: 1 package finished [1.41s]

$ colcon test-result
Summary: 16 tests, 0 errors, 0 failures, 2 skipped
```

danielcranston, 2025-05-02:
> TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

I figure I'd do this as well, so I [added a gtest executable](https://github.com/danielcranston/ros2_testing_repo/commit/2bb18a82f514181398352a3300eb56f35d644af9) that links against `geometry_msgs`. Surprisingly that does not build:

```
$ colcon build
Starting >>> kilted_tutorial_party_340
--- stderr: kilted_tutorial_party_340                             
/home/daniel/colcon_ws/src/ros2_testing_repo/kilted_tutorial_party_340/test/tutorial_test.cpp:3:10: fatal error: geometry_msgs/msg/pose.hpp: No such file or directory
    3 | #include <geometry_msgs/msg/pose.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I might be overlooking something though. And yes, the binary archive was sourced before building:

```
$ ros2 pkg prefix geometry_msgs
/home/daniel/ros2_kilted_binary_archive/ros2-linux
```

danielcranston, 2025-05-02:
I found the issue, it was related to `target_link_libraries` usage. After applying [this](https://github.com/danielcranston/ros2_testing_repo/commit/5621d982eb484fb54bd2a67becf7d4e7357ad1d2): 

```diff
-  target_link_libraries(${PROJECT_NAME}_tutorial_test geometry_msgs)
+  target_link_libraries(${PROJECT_NAME}_tutorial_test ${geometry_msgs_TARGETS})
```

I could build and the tests pass.
```
$ colcon build --cmake-args -DBUILD_TESTING=ON
$ colcon test --event-handlers console_cohesion+
...
1: [==========] Running 1 test from 1 test suite.
1: [----------] Global test environment set-up.
1: [----------] 1 test from kilted_tutorial_party_340
1: [ RUN      ] kilted_tutorial_party_340.test_linking_with_geometry_msgs
1: [       OK ] kilted_tutorial_party_340.test_linking_with_geometry_msgs (0 ms)
1: [----------] 1 test from kilted_tutorial_party_340 (0 ms total)
1: 
1: [----------] Global test environment tear-down
1: [==========] 1 test from 1 test suite ran. (0 ms total)
1: [  PASSED  ] 1 test.
...
```

I'm surprised by the change from linking to `geometry_msgs` to `${geometry_msgs_TARGETS}` though, is this intended? Seems pretty intrusive.
* [Jazzy pub/sub tutorial](https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html) says to do `ament_target_dependencies(talker rclcpp std_msgs)`
* [Kilted pub/sub tutorial](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Publisher-And-Subscriber.html) says to do `target_link_libraries(talker PUBLIC rclcpp::rclcpp ${std_msgs_TARGETS})`

I see that `ament_target_dependencies()` has been deprecated (https://github.com/ament/ament_cmake/pull/572), so I understand that the CMake-native `target_link_libraries()` should be used. Still, I'm surprised the target changed along the way.

---

Also a sidenote, if I add `PUBLIC` or `PRIVATE` to the invocation:

```diff
-  target_link_libraries(${PROJECT_NAME}_tutorial_test ${geometry_msgs_TARGETS})
+  target_link_libraries(${PROJECT_NAME}_tutorial_test PUBLIC ${geometry_msgs_TARGETS})
```

then building starts to fail. I gather this is related to the `ament_target_dependencies()` deprecation.

```
$ colcon build --cmake-args -DBUILD_TESTING=ON
Starting >>> kilted_tutorial_party_340
--- stderr: kilted_tutorial_party_340                         
CMake Error at CMakeLists.txt:44 (target_link_libraries):
  The plain signature for target_link_libraries has already been used with
  the target "kilted_tutorial_party_340_tutorial_test".  All uses of
  target_link_libraries with a target must be either all-keyword or
  all-plain.

  The uses of the plain signature are here:

   * /home/daniel/ros2_kilted_binary_archive/ros2-linux/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:53 (target_link_libraries)
   * /home/daniel/ros2_kilted_binary_archive/ros2-linux/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:55 (target_link_libraries)
   * /home/daniel/ros2_kilted_binary_archive/ros2-linux/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:59 (target_link_libraries)
```

So it seems to me the deprecation and migration away from `ament_target_dependencies()` still has some rough edges, as the Kilted pub/sub tutorial suggests adding `PUBLIC`, but that would be malformed if the target (first argument) was created via `ament_add_gtest_executable()`.

Then again, [the Kilted GTest tutorial](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Testing/Cpp.html#cmakelists-txt) omits adding `PUBLIC`, probably very deliberately so.

danielcranston, 2025-05-12:
This can probably be closed, since the the local workspace overlay works.

A summary of the comments above:
* No longer linking to `<message_package>`, but instead to `${<message_package>_TARGETS}`, was surprising.
* `ament_add_gtest_executable` and `target_link_libraries` gotcha: https://github.com/ament/ament_cmake/issues/405

(Note that these points don't necessarily relate to this ticket)

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde