ros2/kilted_tutorial_party #341: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hi, I worked on this ticket

I built bondcpp from source (https://github.com/ros/bond_core) which depends on rclcpp and rclcpp_lifecycle.

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# rm -rf custom_ws/build/ custom_ws/install/ custom_ws/log/
root@cluttered:~# cd custom_ws/
root@cluttered:~/custom_ws# colcon build --symlink-install
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [0.93s]                                                  
Finished <<< bond [4.32s]                    
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [0.91s]                                             
Finished <<< bondcpp [6.55s]                      
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.60s]                                                        
Finished <<< test_bond [7.88s]                      

Summary: 6 packages finished [18.9s]
root@cluttered:~/custom_ws# source install/setup.bash
root@cluttered:~/custom_ws# colcon test
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [1.73s]                                    
Finished <<< bond [5.42s]          
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondcpp [1.64s]                                      
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.01s]
Finished <<< test_bond [12.5s]                                         
--- stderr: bondpy                     

=============================== warnings summary ===============================
test/test_flake8.py: 16 warnings
  Warning: This process (pid=1141900) is multi-threaded, use of fork() may lead to deadlocks in the child.

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< bondpy [21.3s]

Summary: 6 packages finished [26.8s]
  1 package had stderr output: bondpy
root@cluttered:~/custom_ws# colcon test-result
Summary: 262 tests, 0 errors, 0 failures, 60 skipped
root@cluttered:~/custom_ws# uname -m
x86_64
root@cluttered:~/custom_ws# 
root@cluttered:~/custom_ws# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~/custom_ws# 

```

It works.

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde