ros2/kilted_tutorial_party #342: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-01:
PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-24-generic-aarch64-with-glibc2.39
release          : 6.11.0-24-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

**OUTPUT**

I tested local workspace overlay with ```dual_laser_merger``` package source build. 

It is working as expected and this test passed.

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: FASTDDS
BuildType: debian: pre-built debian packages from the apt repository
OS: Ubuntu Noble
Chip: Arm64
```

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde