ros2/kilted_tutorial_party #343: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-07:
- [ ] ~**Local workspace overlay works**~

Using the same setup as in https://github.com/ros2/kilted_tutorial_party/issues/296#issuecomment-2860575648.

I could build some dummy packages and run the standard tests, **but there was one failure** (test_flake8).

```
# cd src
# ros2 pkg create --build-type ament_cmake my_pkg
# ros2 pkg create --build-type ament_python my_pkg_py
# cd ..
# colcon build --symlink-install --cmake-args -DBUILD_TESTING=ON --packages-select my_pkg my_pkg_py

# colcon test --packages-select my_pkg my_pkg_py
Starting >>> my_pkg_py
Finished <<< my_pkg [3.45s]                                         
Finished <<< my_pkg_py [4.84s]  [ with test failures ]

Summary: 2 packages finished [5.57s]
  1 package had test failures: my_pkg_py

# colcon test-result --verbose
build/my_pkg_py/pytest.xml: 4 tests, 0 errors, 1 failure, 1 skipped
- my_pkg_py.test.test_flake8 test_flake8
  <<< failure message
    ValueError: 'string' is not callable
  >>>

Summary: 8 tests, 0 errors, 1 failure, 1 skipped
```

---

> TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

I extended the above packages to host a `geometry_msgs/Pose` talker/listener, in the same way as https://github.com/ros2/kilted_tutorial_party/issues/329#issuecomment-2853409113.

Terminal 1
```
# . ~/ros2_kilted/install/local_setup.bash
# . install/setup.bash
# ros2 run my_pkg talker 
[INFO] [1746656727.017470731] [talker]: Publishing
[INFO] [1746656727.517663650] [talker]: Publishing
[INFO] [1746656728.017549771] [talker]: Publishing
[INFO] [1746656728.517611334] [talker]: Publishing
```

Terminal 2
```
# . ~/ros2_kilted/install/local_setup.bash
# . install/setup.bash
# ros2 run test_connext_pkg_py listener 
[INFO] [1746656777.596763162] [pose_listener_node]: I heard: geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=1.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=1.0, y=0.0, z=0.0, w=1.0))
[INFO] [1746656778.027018907] [pose_listener_node]: I heard: geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=1.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=1.0, y=0.0, z=0.0, w=1.0))
[INFO] [1746656778.529184804] [pose_listener_node]: I heard: geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=1.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=1.0, y=0.0, z=0.0, w=1.0))
```

danielcranston, 2025-05-12:
Here's the full traceback

<details>
  <summary>Click me</summary>

```
# python3 -m pytest test_flake8.py 
================================================================= test session starts ==================================================================
platform linux -- Python 3.9.21, pytest-6.2.2, py-1.10.0, pluggy-0.13.1
rootdir: /root/ws/src/my_pkg_py
plugins: launch-testing-ros-0.28.1, launch-pytest-3.8.1, launch-testing-3.8.1, ament-xmllint-0.19.2, ament-pep257-0.19.2, ament-mypy-0.19.2, ament-flake8-0.19.2, ament-copyright-0.19.2, ament-lint-0.19.2, repeat-0.9.3, cov-4.0.0, mock-3.6.1, timeout-1.4.2, colcon-core-0.19.0, rerunfailures-9.1.1
collected 1 item                                                                                                                                       

test_flake8.py F                                                                                                                                 [100%]

======================================================================= FAILURES =======================================================================
_____________________________________________________________________ test_flake8 ______________________________________________________________________

    @pytest.mark.flake8
    @pytest.mark.linter
    def test_flake8():
>       rc, errors = main_with_errors(argv=[])

test_flake8.py:22: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
/root/ros2_kilted/build/ament_flake8/ament_flake8/main.py:88: in main_with_errors
    report = generate_flake8_report(
/root/ros2_kilted/build/ament_flake8/ament_flake8/main.py:267: in generate_flake8_report
    style = get_flake8_style_guide(flake8_argv)
/root/ros2_kilted/build/ament_flake8/ament_flake8/main.py:186: in get_flake8_style_guide
    application.plugins, application.options = parse_args(argv)
/root/.local/lib/python3.9/site-packages/flake8/options/parse_args.py:51: in parse_args
    option_manager.register_plugins(plugins)
/root/.local/lib/python3.9/site-packages/flake8/options/manager.py:259: in register_plugins
    add_options(self)
/usr/lib/python3.9/site-packages/flake8_import_order/flake8_linter.py:44: in add_options
    register_opt(
/usr/lib/python3.9/site-packages/flake8_import_order/flake8_linter.py:98: in register_opt
    parser.add_option(*args, **kwargs)
/root/.local/lib/python3.9/site-packages/flake8/options/manager.py:281: in add_option
    self._current_group.add_argument(*option_args, **option_kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <argparse._ArgumentGroup object at 0x70952a4af220>, args = ('--import-order-style',)
kwargs = {'default': 'cryptography', 'dest': 'import_order_style', 'help': 'Style to follow. Available: appnexus, cryptography, edited, google, pep8, pycharm, smarkets', 'option_strings': ['--import-order-style'], ...}
chars = '-', action_class = <class 'argparse._StoreAction'>
action = _StoreAction(option_strings=['--import-order-style'], dest='import_order_style', nargs=None, const=None, default='cryp...alse, help='Style to follow. Available: appnexus, cryptography, edited, google, pep8, pycharm, smarkets', metavar=None)
type_func = 'string'

    def add_argument(self, *args, **kwargs):
        """
        add_argument(dest, ..., name=value, ...)
        add_argument(option_string, option_string, ..., name=value, ...)
        """
    
        # if no positional args are supplied or only one is supplied and
        # it doesn't look like an option string, parse a positional
        # argument
        chars = self.prefix_chars
        if not args or len(args) == 1 and args[0][0] not in chars:
            if args and 'dest' in kwargs:
                raise ValueError('dest supplied twice for positional argument')
            kwargs = self._get_positional_kwargs(*args, **kwargs)
    
        # otherwise, we're adding an optional argument
        else:
            kwargs = self._get_optional_kwargs(*args, **kwargs)
    
        # if no default was supplied, use the parser-level default
        if 'default' not in kwargs:
[... 12 lines cut ...]
        # raise an error if the action type is not callable
        type_func = self._registry_get('type', action.type, action.type)
        if not callable(type_func):
>           raise ValueError('%r is not callable' % (type_func,))
E           ValueError: 'string' is not callable

/usr/lib64/python3.9/argparse.py:1428: ValueError
=============================================================== short test summary info ================================================================
FAILED test_flake8.py::test_flake8 - ValueError: 'string' is not callable
================================================================== 1 failed in 0.50s ===================================================================
```

</details>

This could be related: https://github.com/PyCQA/flake8/issues/1751.

```
# python3 --version
Python 3.9.21
# dnf list | grep pytest
python3-pytest.noarch                                                                    6.2.2-7.el9                          @appstream          
...
python3.11-pytest.noarch                                                                 7.2.0-1.el9                          crb                
python3.12-pytest.noarch                                                                 7.4.2-2.el9                          crb      
```

Aniketkumarroy, 2025-05-14:
Hi, I tested this on `rockylinux:9` here are the results
### **creating cpp and python package in the workspace**

[image]
### **building the packages**

[image]
### **running the package nodes**

[image]

danielcranston, 2025-05-14:
Mind trying again with this instead (like the deprecation warning suggests)?
```
target_link_libraries(pose2d_publisher_node rclcpp::rclcpp ${geometry_msgs_TARGETS})
```

Nice to see the flake8 test passing on your end.

Aniketkumarroy, 2025-05-14:
@danielcranston thanks for that, I used this now the package is building normally as intended but their is still no improvement in running tests

[image]
while running the flake8 test I too got the same results

[image]