ros2/kilted_tutorial_party #344: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Innomer, 2025-05-07:
## Checks
- [x] Local workspace overlay works

## Steps
- Created a new workspace
- Created a new package named dummy_ros2_linker
- Added dependencies to rclcpp and std_msgs
- Created a new CPP Publisher node
- Built the package and workspace
- ros2 run the node

## Proof
[image]

## Result
SUCCESS

ahcorde, 2025-05-07:
Thank you for testing

closed 2025-05-07 by ahcorde