ros2/kilted_tutorial_party #345: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

RishikesavanRamesh, 2025-05-02:
I m interested

akky20, 2025-05-07:
Hey @RishikesavanRamesh , are you working on this issue , or I can test it !!

akky20, 2025-05-07:
## Running Rmw: Zenoh

[image]

## Buidling and Sources 

[image]

[image]

## Testing 

[image]

ahcorde, 2025-05-07:
Thank you for testing