ros2/kilted_tutorial_party #347: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-03:
This feels RMW agnostic, so maybe a duplicate of https://github.com/ros2/kilted_tutorial_party/issues/340? Same goes for the other tickets with same name but different RMW label.

danielcranston, 2025-05-08:
FWIW I gave this a spin.

- [x]  **Local workspace overlay works**

```
$ docker run --rm=true -ti ghcr.io/sloretz/ros-testing:kilted-desktop bash
root@4ebb92ad1eb8:/# dpkg -l | grep ros-kilted-desktop
ii  ros-kilted-desktop                                0.12.0-2noble.20250505.192935       arm64
root@f7f31734f374:/# source /opt/ros/kilted/setup.bash
root@f7f31734f374:/# export RMW_IMPLEMENTATION=rmw_zenoh_cpp
```

```
root@f7f31734f374:/tmp/ws/src# ros2 pkg create --build-type ament_cmake my_test_pkg --node-name talker --dependencies rclcpp geometry_msgs
root@f7f31734f374:/tmp/ws/src# ros2 pkg create --build-type ament_python my_test_pkg_py --node-name listener --dependencies rclpy geometry_msgs
root@f7f31734f374:/tmp/ws/src# cd ..
root@f7f31734f374:/tmp/ws# colcon build            
root@f7f31734f374:/tmp/ws# colcon test
root@f7f31734f374:/tmp/ws# colcon test-result
Summary: 12 tests, 0 errors, 0 failures, 2 skipped
```

> TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

I did the same thing as explained in https://github.com/ros2/kilted_tutorial_party/issues/329#issuecomment-2853409113 and the talker/listener worked as expected (after first starting the Zenoh router in a 3rd terminal, of course).

danielcranston, 2025-05-12:
This can be closed.

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde