ros2/kilted_tutorial_party #348: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hi, I worked on this ticket

I built bondcpp from source (https://github.com/ros/bond_core) which depends on rclcpp and rclcpp_lifecycle.

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_zenoh_cpp
root@cluttered:~# rm -rf custom_ws/build/ custom_ws/install/ custom_ws/log/
root@cluttered:~# cd custom_ws/
root@cluttered:~/custom_ws# colcon build --symlink-install                                     
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [0.91s]                                                  
Finished <<< bond [3.98s]                    
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [0.89s]                                                 
Finished <<< bondcpp [6.32s]                      
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.57s]                                                        
Finished <<< test_bond [7.71s]                      

Summary: 6 packages finished [18.1s]
root@cluttered:~/custom_ws# colcon test                   
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [1.68s]                                    
Finished <<< bond [5.17s]          
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondcpp [1.63s]                                      
Starting >>> bond_core
Starting >>> test_bond
Finished <<< bond_core [0.01s]                                                         
Finished <<< test_bond [16.3s]                                         
--- stderr: bondpy                     

=============================== warnings summary ===============================
test/test_flake8.py: 16 warnings
  Warning: This process (pid=1135865) is multi-threaded, use of fork() may lead to deadlocks in the child.

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< bondpy [23.5s]

Summary: 6 packages finished [28.7s]
  1 package had stderr output: bondpy
root@cluttered:~/custom_ws# colcon test-result
Summary: 262 tests, 0 errors, 0 failures, 60 skipped
root@cluttered:~/custom_ws# uname -m
x86_64
```

It works.

ahcorde, 2025-05-02:
Thank you for testing