ros2/kilted_tutorial_party #35: Check ROS 2 daemon CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check status**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>
- [ ] **Stop the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon stop
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been stopped
       ```
  </details>
- [ ] **Check that the daemon is not running**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is not running
       ```
  </details>
- [ ] **Restart the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon start
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been started
       ```
  </details>
- [ ] **Check restart**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-02:
Picking up this one as well!

Check status:
'The daemon is running'

Stop the daemon:
'The daemon has been stopped'

Check that the daemon is not running:
'The daemon is not running'

Restart the daemon:
'The daemon has been started'

Check restart:
'The daemon is running'

All tests pass. ready to be closed!

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde