ros2/kilted_tutorial_party #352: macOS Development Setup

tfoote, 2025-05-01:
Check the documentation for the 'macOS Development Setup' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [macOS Development Setup page](https://docs.ros.org/en/kilted/Installation/Alternatives/macOS-Development-Setup.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-04:
This one doesn't really make sense, does it ?

JatinVira, 2025-05-13:
@ahcorde 
The tag on this needs to be changed from `noble` to `mac` or something more appropiate

mjcarroll, 2025-05-16:
This needs to be removed entirely.

closed 2025-05-16 by mjcarroll

JatinVira, 2025-05-16:
@mjcarroll 
Can this issue have an assigne?

Thanks