ros2/kilted_tutorial_party #354: Managing nodes with managed lifecycles

tfoote, 2025-05-01:
Check the documentation for the 'Managing nodes with managed lifecycles' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Managing nodes with managed lifecycles page](https://docs.ros.org/en/kilted/Tutorials/Demos/Managed-Nodes.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

NickTziaros, 2025-05-02:
## ❓ 

This is page leading to Lifecycle demo #324.
I don't think the checks are applicable.

This page has always seemed a bit unclear to me. I'm currently working on a tutorial about lifecycle nodes ([#4924](https://github.com/ros2/ros2_documentation/issues/4924)), so once that's finished, we might want to revisit this page and discuss whether it's still necessary. But for the purpose of this test, I think it's good to go.

fujitatomoya, 2025-05-13:
I think this issue means that we test https://github.com/ros2/demos/blob/kilted/lifecycle/README.rst instead.

fujitatomoya, 2025-05-13:
```console
root@tomoyafujita:~# ros2 run lifecycle lifecycle_service_client
[INFO] [1747112565.967481988] [lc_client]: Transition 1 successfully triggered.
[INFO] [1747112565.967643999] [lc_client]: Node lc_talker has current state inactive.
[INFO] [1747112577.855636574] [lc_client]: Transition 3 successfully triggered.
[INFO] [1747112577.856387915] [lc_client]: Node lc_talker has current state active.
[INFO] [1747112585.855306783] [lc_client]: Transition 4 successfully triggered.
[INFO] [1747112585.856076482] [lc_client]: Node lc_talker has current state inactive.
[INFO] [1747112597.855694065] [lc_client]: Transition 3 successfully triggered.
[INFO] [1747112597.856635587] [lc_client]: Node lc_talker has current state active.
[INFO] [1747112605.855360264] [lc_client]: Transition 4 successfully triggered.
[INFO] [1747112605.856287346] [lc_client]: Node lc_talker has current state inactive.
[INFO] [1747112615.856209943] [lc_client]: Transition 2 successfully triggered.
[INFO] [1747112615.856875796] [lc_client]: Node lc_talker has current state unconfigured.
[INFO] [1747112625.855318796] [lc_client]: Transition 5 successfully triggered.
[INFO] [1747112625.856103909] [lc_client]: Node lc_talker has current state finalized.

root@tomoyafujita:~# ros2 run lifecycle lifecycle_listener
[INFO] [1747112565.966898596] [lc_listener]: notify callback: Transition from state unconfigured to configuring
[INFO] [1747112565.967423996] [lc_listener]: notify callback: Transition from state configuring to inactive
[INFO] [1747112575.855042610] [lc_listener]: notify callback: Transition from state inactive to activating
[INFO] [1747112577.855427737] [lc_listener]: notify callback: Transition from state activating to active
[INFO] [1747112577.855799705] [lc_listener]: data_callback: Lifecycle HelloWorld #10
[INFO] [1747112577.967988531] [lc_listener]: data_callback: Lifecycle HelloWorld #11
[INFO] [1747112578.967990124] [lc_listener]: data_callback: Lifecycle HelloWorld #12
[INFO] [1747112579.968089965] [lc_listener]: data_callback: Lifecycle HelloWorld #13
[INFO] [1747112580.968138100] [lc_listener]: data_callback: Lifecycle HelloWorld #14
[INFO] [1747112581.968088027] [lc_listener]: data_callback: Lifecycle HelloWorld #15
[INFO] [1747112582.968076079] [lc_listener]: data_callback: Lifecycle HelloWorld #16
[INFO] [1747112583.968067753] [lc_listener]: data_callback: Lifecycle HelloWorld #17
[INFO] [1747112584.968083236] [lc_listener]: data_callback: Lifecycle HelloWorld #18
[INFO] [1747112585.855060099] [lc_listener]: notify callback: Transition from state active to deactivating
[INFO] [1747112585.855223912] [lc_listener]: notify callback: Transition from state deactivating to inactive
[INFO] [1747112595.855145060] [lc_listener]: notify callback: Transition from state inactive to activating
[INFO] [1747112597.855556159] [lc_listener]: notify callback: Transition from state activating to active
[INFO] [1747112597.855753478] [lc_listener]: data_callback: Lifecycle HelloWorld #29
[INFO] [1747112597.968277632] [lc_listener]: data_callback: Lifecycle HelloWorld #30
[INFO] [1747112598.968143128] [lc_listener]: data_callback: Lifecycle HelloWorld #31
[INFO] [1747112599.968045457] [lc_listener]: data_callback: Lifecycle HelloWorld #32
[INFO] [1747112600.968108673] [lc_listener]: data_callback: Lifecycle HelloWorld #33
[INFO] [1747112601.968086234] [lc_listener]: data_callback: Lifecycle HelloWorld #34
[INFO] [1747112602.968069893] [lc_listener]: data_callback: Lifecycle HelloWorld #35
[INFO] [1747112603.968075261] [lc_listener]: data_callback: Lifecycle HelloWorld #36
[INFO] [1747112604.968130674] [lc_listener]: data_callback: Lifecycle HelloWorld #37
[INFO] [1747112605.855043705] [lc_listener]: notify callback: Transition from state active to deactivating
[INFO] [1747112605.855193562] [lc_listener]: notify callback: Transition from state deactivating to inactive
[INFO] [1747112615.855163457] [lc_listener]: notify callback: Transition from state inactive to cleaningup
[INFO] [1747112615.856047945] [lc_listener]: notify callback: Transition from state cleaningup to unconfigured
[INFO] [1747112625.855123309] [lc_listener]: notify callback: Transition from state unconfigured to shuttingdown
[INFO] [1747112625.855266914] [lc_listener]: notify callback: Transition from state shuttingdown to finalized
^C[INFO] [1747112632.104437445] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747112632.104674849] [rclcpp]: signal_handler(signum=2)

root@tomoyafujita:~# ros2 run lifecycle lifecycle_talker
[INFO] [1747112565.967358916] [lc_talker]: on_configure() is called.
[INFO] [1747112566.967581146] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [1747112566.967708572] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[INFO] [1747112567.967584246] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112568.967568504] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112569.967575463] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112570.967585143] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112571.967534002] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[... cut ...]
[INFO] [1747112613.967576840] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112614.967597825] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[INFO] [1747112615.855716098] [lc_talker]: on cleanup is called.
[INFO] [1747112625.854853133] [lc_talker]: on shutdown is called from state unconfigured.
^C[INFO] [1747112631.512082276] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1747112631.512359813] [rclcpp]: signal_handler(signum=2)

```

fujitatomoya, 2025-05-13:
```console
root@tomoyafujita:~# ros2 launch lifecycle lifecycle_demo_launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-12-22-04-35-802003-tomoyafujita-322
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_talker-1]: process started with pid [341]
[INFO] [lifecycle_listener-2]: process started with pid [342]
[INFO] [lifecycle_service_client-3]: process started with pid [343]
[lifecycle_listener-2] [INFO] [1747112676.002282707] [lc_listener]: notify callback: Transition from state unconfigured to configuring
[lifecycle_talker-1] [INFO] [1747112676.002789352] [lc_talker]: on_configure() is called.
[lifecycle_listener-2] [INFO] [1747112676.002861638] [lc_listener]: notify callback: Transition from state configuring to inactive
[lifecycle_service_client-3] [INFO] [1747112676.002924941] [lc_client]: Transition 1 successfully triggered.
[lifecycle_service_client-3] [INFO] [1747112676.003060452] [lc_client]: Node lc_talker has current state inactive.
[lifecycle_talker-1] [INFO] [1747112677.002997934] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [WARN] [1747112677.003117316] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
[lifecycle_talker-1] [INFO] [1747112678.003041489] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112679.003029586] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112680.003026085] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112681.003034211] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112682.003034036] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112683.003035113] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112684.003029807] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112685.003029229] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[lifecycle_talker-1] [INFO] [1747112685.894812134] [lc_talker]: on_activate() is called.
[lifecycle_listener-2] [INFO] [1747112685.895061887] [lc_listener]: notify callback: Transition from state inactive to activating
[lifecycle_talker-1] [INFO] [1747112687.895330968] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #10]
[lifecycle_listener-2] [INFO] [1747112687.895349716] [lc_listener]: notify callback: Transition from state activating to active
[lifecycle_listener-2] [INFO] [1747112687.895795471] [lc_listener]: data_callback: Lifecycle HelloWorld #10
[lifecycle_service_client-3] [INFO] [1747112687.895496558] [lc_client]: Transition 3 successfully triggered.
[lifecycle_service_client-3] [INFO] [1747112687.896320584] [lc_client]: Node lc_talker has current state active.
[lifecycle_talker-1] [INFO] [1747112688.002969553] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #11]
[lifecycle_listener-2] [INFO] [1747112688.003482265] [lc_listener]: data_callback: Lifecycle HelloWorld #11
[lifecycle_talker-1] [INFO] [1747112689.002926515] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #12]
[lifecycle_listener-2] [INFO] [1747112689.003121725] [lc_listener]: data_callback: Lifecycle HelloWorld #12
[lifecycle_talker-1] [INFO] [1747112690.002991867] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #13]
[lifecycle_listener-2] [INFO] [1747112690.003476664] [lc_listener]: data_callback: Lifecycle HelloWorld #13
[lifecycle_talker-1] [INFO] [1747112691.003066338] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #14]
[lifecycle_listener-2] [INFO] [1747112691.003309126] [lc_listener]: data_callback: Lifecycle HelloWorld #14
[lifecycle_talker-1] [INFO] [1747112692.003013093] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #15]
[lifecycle_listener-2] [INFO] [1747112692.003512960] [lc_listener]: data_callback: Lifecycle HelloWorld #15
[lifecycle_talker-1] [INFO] [1747112693.003008395] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #16]
[lifecycle_listener-2] [INFO] [1747112693.003493021] [lc_listener]: data_callback: Lifecycle HelloWorld #16
[lifecycle_talker-1] [INFO] [1747112694.003033976] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #17]
[lifecycle_listener-2] [INFO] [1747112694.003551309] [lc_listener]: data_callback: Lifecycle HelloWorld #17
[lifecycle_talker-1] [INFO] [1747112695.003003395] [lc_talker]: Lifecycle publisher is active. Publishing: [Lifecycle HelloWorld #18]
[lifecycle_listener-2] [INFO] [1747112695.003485773] [lc_listener]: data_callback: Lifecycle HelloWorld #18
[lifecycle_talker-1] [INFO] [1747112695.894779802] [lc_talker]: on_deactivate() is called.
[lifecycle_listener-2] [INFO] [1747112695.895042173] [lc_listener]: notify callback: Transition from state active to deactivating
[lifecycle_listener-2] [INFO] [1747112695.895191020] [lc_listener]: notify callback: Transition from state deactivating to inactive
[... cut ...]
[INFO] [launch]: process[lifecycle_service_client-3] was required: shutting down launched system
[INFO] [lifecycle_listener-2]: sending signal 'SIGINT' to process[lifecycle_listener-2]
[INFO] [lifecycle_talker-1]: sending signal 'SIGINT' to process[lifecycle_talker-1]
[lifecycle_listener-2] [INFO] [1747112736.905288040] [rclcpp]: signal_handler(signum=2)
[lifecycle_talker-1] [INFO] [1747112736.905494465] [rclcpp]: signal_handler(signum=2)
[INFO] [lifecycle_listener-2]: process has finished cleanly [pid 342]
[INFO] [lifecycle_talker-1]: process has finished cleanly [pid 341]

```

fujitatomoya, 2025-05-13:
demo apps work without any problems.

closed 2025-05-13 by fujitatomoya