ros2/kilted_tutorial_party #357: Monitoring For Parameter Changes Python

tfoote, 2025-05-01:
Check the documentation for the 'Monitoring For Parameter Changes Python' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Monitoring For Parameter Changes Python page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-Python.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-04:
- [x] **I was able to follow the documentation.**
- [x] **The documentation seemed clear to me.**
- [x] **The documentation didn't have any obvious errors.**

Monitoring own node's parameter:
```bash
$ ros2 run python_parameter_event_handler node_with_parameters
[INFO] [1746325320.882409862] [node_with_parameters]: Received an update to parameter: an_int_param: 43
```

Monitoring another node's parameter:
```bash
$ ros2 run python_parameter_event_handler node_with_parameters
[INFO] [1746325716.169610910] [node_with_parameters]: Received an update to parameter: a_double_param: 3.45
```

Some comments:

* In the original pasted Python code, specifying `node_name="node_with_parameters"` to `self.handler.add_parameter_callback` is redundant.
* When instructed to extend the original code to add a parameter to another node's parameters, the instruction makes the user re-assign `self.callback_handle`, which probably isn't ideal.
* It would be really nice if there was a link to the `ParameterEventHandler` doc: https://docs.ros.org/en/rolling/p/rclpy/rclpy.parameter_event_handler.html. But it seems empty, I'm not sure what's up with that.
* Another really useful function of `ParameterEventHandler` is [add_parameter_event_callback](https://github.com/ros2/rclpy/blob/kilted/rclpy/rclpy/parameter_event_handler.py#L147C13-L147C41). Maybe it should be mentioned.

danielcranston, 2025-05-12:
PR is up and awaiting a second review from @fujitatomoya.

This can probably be closed, unless we want to keep it open until the PR is merged.

One outstanding question though:

> It would be really nice if there was a link to the `ParameterEventHandler` doc: https://docs.ros.org/en/rolling/p/rclpy/rclpy.parameter_event_handler.html. But it seems empty, I'm not sure what's up with that?

closed 2025-05-13 by fujitatomoya