ros2/kilted_tutorial_party #359: Parameter executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-05:
I can take this one!

Terminal 1:
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1746471049.194907100] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
```

Terminal 2:
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run demo_nodes_py async_param_client
[INFO] [1746471073.866329700] [async_param_client]: Setting parameters:
[INFO] [1746471073.887392400] [async_param_client]:     int_parameter:
[INFO] [1746471073.895643200] [async_param_client]:         successful: True
[INFO] [1746471073.902852600] [async_param_client]:     bool_parameter:
[INFO] [1746471073.910383400] [async_param_client]:         successful: True
[INFO] [1746471073.916287500] [async_param_client]:     string_parameter:
[INFO] [1746471073.920485300] [async_param_client]:         successful: True
[INFO] [1746471073.927102700] [async_param_client]: Listing Parameters:
[INFO] [1746471073.943951300] [async_param_client]:     - bool_parameter
[INFO] [1746471073.951352100] [async_param_client]:     - int_parameter
[INFO] [1746471073.968451300] [async_param_client]:     - string_parameter
[INFO] [1746471073.975025700] [async_param_client]: Getting parameters:
[INFO] [1746471073.989793800] [async_param_client]:     - int_parameter: 10
[INFO] [1746471073.995713300] [async_param_client]:     - bool_parameter: False
[INFO] [1746471074.002877800] [async_param_client]:     - string_parameter: Fee Fi Fo Fum
[INFO] [1746471074.014695300] [async_param_client]: Loading parameters:
[INFO] [1746471074.169035900] [async_param_client]:     other_int_parameter:
[INFO] [1746471074.190952400] [async_param_client]:         successful: True
[INFO] [1746471074.205728300] [async_param_client]:         value: 0
[INFO] [1746471074.212772200] [async_param_client]:     int_parameter:
[INFO] [1746471074.227327600] [async_param_client]:         successful: True
[INFO] [1746471074.234738300] [async_param_client]:         value: 12
[INFO] [1746471074.240295600] [async_param_client]:     string_parameter:
[INFO] [1746471074.248460300] [async_param_client]:         successful: True
[INFO] [1746471074.253795500] [async_param_client]:         value: I smell the blood of an Englishman
[INFO] [1746471074.259952400] [async_param_client]:     other_string_parameter:
[INFO] [1746471074.267405700] [async_param_client]:         successful: True
[INFO] [1746471074.273359200] [async_param_client]:         value: One fish, two fish, Red fish, blue fish
[INFO] [1746471074.279413900] [async_param_client]:     bool_parameter:
[INFO] [1746471074.284752400] [async_param_client]:         successful: True
[INFO] [1746471074.289748200] [async_param_client]:         value: False
[INFO] [1746471074.295162700] [async_param_client]: Deleting parameters:
[INFO] [1746471074.308067700] [async_param_client]:     other_int_parameter:
[INFO] [1746471074.316703300] [async_param_client]:         successful: True
[INFO] [1746471074.323004200] [async_param_client]:         reason:
[INFO] [1746471074.333258600] [async_param_client]:     other_string_parameter:
[INFO] [1746471074.341623600] [async_param_client]:         successful: True
[INFO] [1746471074.349722500] [async_param_client]:         reason:
[INFO] [1746471074.356108300] [async_param_client]:     string_parameter:
[INFO] [1746471074.362498600] [async_param_client]:         successful: True
[INFO] [1746471074.370117000] [async_param_client]:         reason:
```

Response is identical to with fastdds. Tests passed! Ready to close.

Yadunund, 2025-05-05:
Thanks for your contribution!

closed 2025-05-05 by Yadunund