ros2/kilted_tutorial_party #36: Check ROS 2 daemon CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check status**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>
- [ ] **Stop the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon stop
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been stopped
       ```
  </details>
- [ ] **Check that the daemon is not running**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is not running
       ```
  </details>
- [ ] **Restart the daemon**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon start
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon has been started
       ```
  </details>
- [ ] **Check restart**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 daemon status
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       The daemon is running
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

rtarun1, 2025-05-02:
## Setup 

- Rmw: Fastdds
- BuildType: Debian
- Chip: Amd64
- Inside Docker container running `ghcr.io/sloretz/ros-testing:kilted-desktop`

## Output 

- [x]  **Check status**
- [x]  **Stop the daemon**
- [x]  **Check that the daemon is not running**
- [x]  **Restart the daemon**
- [x]  **Check restart**

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde