ros2/kilted_tutorial_party #360: Parameter executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
It works for me

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde