ros2/kilted_tutorial_party #361: Parameter executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-02:
I can take this one as well!

JasperGrant, 2025-05-02:
Parameter Blackboard:
`[INFO] [1746197626.794650100] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!`

Async Parameter Client:
`[INFO] [1746197642.728011600] [async_param_client]: Setting parameters:
[INFO] [1746197642.731364600] [async_param_client]:     int_parameter:
[INFO] [1746197642.732992700] [async_param_client]:         successful: True
[INFO] [1746197642.734375300] [async_param_client]:     bool_parameter:
[INFO] [1746197642.735728600] [async_param_client]:         successful: True
[INFO] [1746197642.737525100] [async_param_client]:     string_parameter:
[INFO] [1746197642.738970700] [async_param_client]:         successful: True
[INFO] [1746197642.740457700] [async_param_client]: Listing Parameters:
[INFO] [1746197642.743362900] [async_param_client]:     - bool_parameter
[INFO] [1746197642.745030100] [async_param_client]:     - int_parameter
[INFO] [1746197642.746952100] [async_param_client]:     - string_parameter
[INFO] [1746197642.748564100] [async_param_client]: Getting parameters:
[INFO] [1746197642.751037300] [async_param_client]:     - int_parameter: 10
[INFO] [1746197642.752772900] [async_param_client]:     - bool_parameter: False
[INFO] [1746197642.754369300] [async_param_client]:     - string_parameter: Fee Fi Fo Fum
[INFO] [1746197642.756055300] [async_param_client]: Loading parameters:
[INFO] [1746197642.763896300] [async_param_client]:     other_int_parameter:
[INFO] [1746197642.765815400] [async_param_client]:         successful: True
[INFO] [1746197642.767671100] [async_param_client]:         value: 0
[INFO] [1746197642.769441600] [async_param_client]:     int_parameter:
[INFO] [1746197642.771214500] [async_param_client]:         successful: True
[INFO] [1746197642.773453400] [async_param_client]:         value: 12
[INFO] [1746197642.775629300] [async_param_client]:     string_parameter:
[INFO] [1746197642.777711100] [async_param_client]:         successful: True
[INFO] [1746197642.779834300] [async_param_client]:         value: I smell the blood of an Englishman
[INFO] [1746197642.781911800] [async_param_client]:     other_string_parameter:
[INFO] [1746197642.784054500] [async_param_client]:         successful: True
[INFO] [1746197642.786074100] [async_param_client]:         value: One fish, two fish, Red fish, blue fish
[INFO] [1746197642.788301300] [async_param_client]:     bool_parameter:
[INFO] [1746197642.790508800] [async_param_client]:         successful: True
[INFO] [1746197642.794385100] [async_param_client]:         value: False
[INFO] [1746197642.796643400] [async_param_client]: Deleting parameters:
[INFO] [1746197642.799836600] [async_param_client]:     other_int_parameter:
[INFO] [1746197642.802481600] [async_param_client]:         successful: True
[INFO] [1746197642.804912300] [async_param_client]:         reason:
[INFO] [1746197642.807391800] [async_param_client]:     other_string_parameter:
[INFO] [1746197642.810208800] [async_param_client]:         successful: True
[INFO] [1746197642.812522800] [async_param_client]:         reason:
[INFO] [1746197642.814892900] [async_param_client]:     string_parameter:
[INFO] [1746197642.817548900] [async_param_client]:         successful: True
[INFO] [1746197642.819934100] [async_param_client]:         reason:
`
Test passed!

closed 2025-05-12 by ahcorde