ros2/kilted_tutorial_party #362: Parameter executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Async parameter client**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp parameter_blackboard
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py async_param_client
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-02:
I am interested.

Tenveis, 2025-05-02:
Following are the results:

*    I was able to follow the documentation
*    The documentation was clear
*    I didn't find any errors in the documentation

- [x]  Async parameter client

[image]

fujitatomoya, 2025-05-09:
@Tenveis thanks for testing, i will go ahead to close this issue.

closed 2025-05-09 by fujitatomoya