ros2/kilted_tutorial_party #364: Pluginlib

tfoote, 2025-05-01:
Check the documentation for the 'Pluginlib' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Pluginlib page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Pluginlib.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-03:
hello, trying this

dheerubhai-101, 2025-05-03:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

Finding a similar issue with `target_link_libraries`, where colcon build throws an error of not finding the header.

```
vscode ➜ /workspaces/kilted_ws $ colcon build --packages-select polygon_base polygon_plugins
Starting >>> polygon_base
Finished <<< polygon_base [0.08s]
Starting >>> polygon_plugins
--- stderr: polygon_plugins                         
/workspaces/kilted_ws/src/polygon_plugins/src/polygon_plugins.cpp:1:10: fatal error: polygon_base/regular_polygon.hpp: No such file or directory
    1 | #include <polygon_base/regular_polygon.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/polygon_plugins.dir/build.make:76: CMakeFiles/polygon_plugins.dir/src/polygon_plugins.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/polygon_plugins.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< polygon_plugins [0.29s, exited with code 2]

Summary: 1 package finished [0.47s]
  1 package failed: polygon_plugins
  1 package had stderr output: polygon_plugins
```

On replacing this with `ament_target_dependencies`, package builds but with deprecation warning.Trying to figure out why the header is not being visible although header exists in the install directory.

[image]

ahcorde, 2025-05-09:
The version `0.38` should have this fix, Related PR https://github.com/ros2/ros2cli/issues/973

dheerubhai-101, 2025-05-11:
built kilted with source and replaced ros2cli with `0.38` release, but still having issue building the `polygon_plugins` package. It still is not able to find the header file.

JatinVira, 2025-05-12:
Hey @ahcorde 
I independently tried out the same on my machine using the latest docker image and following the same set of steps as mentioned on the official documentation attached, and hit the same roadblock as @dheerubhai-101 described above

[image]

My system configuration

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-139-generic-x86_64-with-glibc2.39
release          : 5.15.0-139-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

This is an issue that needs to be fixed.

dheerubhai-101, 2025-05-13:
Hi, that will be the case as mentioned earlier. But we would need `target_link_libraries` in CmakeLists.txt going ahead

Tanishq30052002, 2025-05-13:
[image]
Output

clalancette, 2025-05-13:
> i will have a try

You can have a look at https://github.com/ros2/realtime_support/blob/rolling/tlsf_cpp/CMakeLists.txt and https://github.com/ros2/demos/blob/rolling/pendulum_control/CMakeLists.txt for how this is done with the `tlsf_cpp` target.

Tanishq30052002, 2025-05-13:
> > i will have a try
> 
> You can have a look at https://github.com/ros2/realtime_support/blob/rolling/tlsf_cpp/CMakeLists.txt and https://github.com/ros2/demos/blob/rolling/pendulum_control/CMakeLists.txt for how this is done with the `tlsf_cpp` target.

[image]

Problem Resolved, I will create a PR for the smame

Tanishq30052002, 2025-05-13:
@ahcorde can you please have a look over the PR
Thanks

closed 2025-05-14 by ahcorde

Tanishq30052002, 2025-05-14:
@ahcorde can you assign me as the assigne