ros2/kilted_tutorial_party #367: Quality of Service Demos

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Demo URL](https://github.com/ros2/demos/blob/kilted/quality_of_service_demo/README.md)

## Checks
- [ ] **Deadline with pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
       ```
  
    **Expect**
    
    1.
       The publisher will publish for 5 seconds, then pause for 2 - deadline events will start firing on both participants, until the publisher comes back.
    
  </details>
- [ ] **Deadline no pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 0
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 0
       ```
  
    **Expect**
    
    1.
       The publisher doesn't actually pause, no deadline misses should occur.
    
  </details>
- [ ] **Lifespan, some survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see (approximately) messages 4-9 printed from the subscriber.
       
       The first few messages, with 1 second lifespan, were gone by the time the subscriber joined after 3 seconds.
       
    
  </details>
- [ ] **Lifespan, all survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see all of the messages (0-9) printed from the subscriber.
       
       All messages, with their 4 second lifespan, survived until the subscriber joined.
       
    
  </details>
- [ ] **Liveliness**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000
       ```
  
    **Expect**
    
    1.
       After 2 seconds, the publisher will be killed, and the subscriber will receive a callback 1 second after that notifying it that the liveliness has changed.
    
  </details>
- [ ] **Incompatible QoS Offered/Requested**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp incompatible_qos durability
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py incompatible_qos durability
       ```
  
    **Expect**
    
    1.
       The publisher in this demo will publish 5 messages, but also output a notification stating that its offered QoS is incompatible with the subscriber that is also created by this demo. The subscriber in this demo will similarly output a notification stating that its requested QoS is incompatible with the QoS offered by the publisher.
       
    
  </details>
- [ ] **Interactive Quality of Service Demo**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp interactive_publisher
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run quality_of_service_demo_cpp interactive_subscriber
       ```
    1.
       In the publisher's terminal, press `p` to manually assert the liveliness of the publisher, for when Liveliness kind is `MANUAL_BY_TOPIC`.
       
    1.
       In the publisher's terminal, press `s` to toggle enabling/disabling of publishing messages
       
    1.
       In the publisher and subscriber's terminal, press `q` to print the current QoS policy.
       
    1.
       In the publisher and subscriptions's terminal, press `x` to stop and exit the demo.
       
  
    **Expect**
    
    1.
       All key-press commands work.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-03:
I'm testing this

khaledgabr77, 2025-05-03:
I have completed the following tests and confirmed everything works as expected.

## Checks

* [x]  **Deadline with pause**   

[image]

* [x]  **Deadline no pause**   

[image]

* [x]  **Lifespan, some survive**   

[image]

* [x]  **Lifespan, all survive**

[image]

* [x]  **Lifespan, all survive**

[image]

* [x]  **Incompatible QoS Offered/Requested**

[image]

* [x]  **Interactive Quality of Service Demo**

[image]

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde