ros2/kilted_tutorial_party #368: Quality of Service Demos

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Demo URL](https://github.com/ros2/demos/blob/kilted/quality_of_service_demo/README.md)

## Checks
- [ ] **Deadline with pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 2000
       ```
  
    **Expect**
    
    1.
       The publisher will publish for 5 seconds, then pause for 2 - deadline events will start firing on both participants, until the publisher comes back.
    
  </details>
- [ ] **Deadline no pause**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp deadline 600 --publish-for 5000 --pause-for 0
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py deadline 600 --publish-for 5000 --pause-for 0
       ```
  
    **Expect**
    
    1.
       The publisher doesn't actually pause, no deadline misses should occur.
    
  </details>
- [ ] **Lifespan, some survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 1000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see (approximately) messages 4-9 printed from the subscriber.
       
       The first few messages, with 1 second lifespan, were gone by the time the subscriber joined after 3 seconds.
       
    
  </details>
- [ ] **Lifespan, all survive**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py lifespan 4000 --publish-count 10 --subscribe-after 3000
       ```
  
    **Expect**
    
    1.
       After a few seconds, you should see all of the messages (0-9) printed from the subscriber.
       
       All messages, with their 4 second lifespan, survived until the subscriber joined.
       
    
  </details>
- [ ] **Liveliness**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp liveliness 1000 --kill-publisher-after 2000
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py liveliness 1000 --kill-publisher-after 2000
       ```
  
    **Expect**
    
    1.
       After 2 seconds, the publisher will be killed, and the subscriber will receive a callback 1 second after that notifying it that the liveliness has changed.
    
  </details>
- [ ] **Incompatible QoS Offered/Requested**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp incompatible_qos durability
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_py incompatible_qos durability
       ```
  
    **Expect**
    
    1.
       The publisher in this demo will publish 5 messages, but also output a notification stating that its offered QoS is incompatible with the subscriber that is also created by this demo. The subscriber in this demo will similarly output a notification stating that its requested QoS is incompatible with the QoS offered by the publisher.
       
    
  </details>
- [ ] **Interactive Quality of Service Demo**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run quality_of_service_demo_cpp interactive_publisher
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run quality_of_service_demo_cpp interactive_subscriber
       ```
    1.
       In the publisher's terminal, press `p` to manually assert the liveliness of the publisher, for when Liveliness kind is `MANUAL_BY_TOPIC`.
       
    1.
       In the publisher's terminal, press `s` to toggle enabling/disabling of publishing messages
       
    1.
       In the publisher and subscriber's terminal, press `q` to print the current QoS policy.
       
    1.
       In the publisher and subscriptions's terminal, press `x` to stop and exit the demo.
       
  
    **Expect**
    
    1.
       All key-press commands work.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed testing for Issue #368 on Windows 11 / amd64 / binary / Fast DDS.

**Results:**
*   **Deadline (with pause):** PASSED. Deadline missed events correctly occurred during the pause for both C++ and Python versions.
*   **Deadline (no pause):** PASSED. No deadline missed events occurred for either C++ or Python version.
*   **Lifespan (some survive):** PASSED. Listener correctly received messages starting from ~#4 for both C++ and Python, matching expectation.
*   **Lifespan (all survive):** PASSED. Listener correctly received all messages (0-9) for both C++ and Python.
*   **Liveliness:** PASSED. Listener correctly detected the publisher loss after the specified timeout for both C++ and Python.
*   **Incompatible QoS:** PASSED. Both publisher and listener reported incompatible QoS notifications as expected for both C++ and Python.
*   **Interactive QoS Demo:** **PARTIAL FAILURE.**
    *   The `s` (toggle sending), `q` (print QoS), and `x` (exit) keys worked correctly in both publisher and subscriber terminals for the C++ version (did not explicitly test Python interactive, but assume similar).
    *   However, pressing the `p` (assert liveliness) key in the *publisher* terminal resulted in a `DATAWRITER Error: Could not assert liveliness of writer...` message, although the node acknowledged the key press by printing "asserting publisher liveliness". This indicates a potential failure with manual liveliness assertion in this specific demo/configuration.

**Conclusion:** All QoS demo checks passed except for the manual liveliness assertion (`p` key) in the interactive demo, which generated a DDS error.

**(Optional: Add terminal output for the interactive demo inside `<details>` tags if you want).**

ShahazadAbdulla, 2025-05-04:
Just re-ran the C++ "Interactive Quality of Service Demo" test specifically with Fast DDS again to be certain.

Confirmed that pressing the `p` key in the `interactive_publisher` terminal still produces the `DATAWRITER Error: Could not assert liveliness...` message, whereas this key worked correctly without error when testing with CycloneDDS for Issue #366.

marlvernchimbwanda, 2025-05-09:
Successfully completed testing for Issue https://github.com/ros2/kilted_tutorial_party/issues/368 on Windows 10 / amd64 / binary / Fast DDS.
**Results:**

1. **Deadline with pause** : PASSED. Deadline missed events correctly occurred during the pause for both C++ and Python versions.
2. **Deadline no pause** : PASSED. No deadline Missed events occcured it was just continous for both python nand C++

- [image]

- [image]

3. **Lifespan, some survive** : PASSED but it stoped publishing at on 9th message.  For both the python and C++ version

[image]

4. **Lifespan, all survive** PASSED: They all survived with messages from 1 to 9 being retreived for both python and C++

[image]

5. **Liveliness**PASSED: Listener detected the loss of the publisher at the specified interval for both python and C++ 

[image]
6. Incomaptible QoS : PASSED . The listener reported incompatible qos and also the talker acknoledged that it offered incomaptible qos 

[image]

7. The s (toggle sending), q (print QoS), and x (exit) keys worked correctly in both publisher and subscriber terminals for the C++ version despite that pressing the `p[` asserting live key was giving an error whic read that  `2025-05-09 12:49:48.386 [DATAWRITER Error] Could not assert liveliness of writer 01.0f.bc.84.74.43.33.71.00.00.00.00|0.0.13.3 -> Function eprosima::fastdds::dds::DataWriterImpl::assert_liveliness   ` but overally all the keys worked both in the publisher and subscriber

[image]

Conclusion: All QoS demo checks passed except for the manual liveliness assertion (p key) in the interactive demo, which generated a DDS error.

ahcorde, 2025-05-15:
Thank you for testing

closed 2025-05-15 by ahcorde