ros2/kilted_tutorial_party #371: Quaternion Fundamentals

tfoote, 2025-05-01:
Check the documentation for the 'Quaternion Fundamentals' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Quaternion Fundamentals page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Quaternion-Fundamentals.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-02:
- [X]  **I was able to follow the documentation.**
- [X]  **The documentation seemed clear to me.**
- [X]  **The documentation didn't have any obvious errors.**

First time reading through https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Quaternion-Fundamentals.html, I hadn't seen it before. I think it's really good and covers the most important information readers might want to know.

Here are some things that came to mind:
* The fact that `geometry_msgs/Quaternion` is default constructed as identity (w=1), but `tf2::Quaternion` is not.
* It's great that the text explains that negating `w` equals inversion. On the same note, it might be good to mention that negating _all elements_ is a "no-op", the result represents exactly the same rotation. I think this is something newcomers to quaternions find surprising. For instance, blindly using `q.w ~= 1.0` as a identity test is dangerous.
* It's clear that the go-to library/class for dealing with quaternions in ROS2 is `tf2::Quaternion` for C++, but what about Python? AFAIK there isn't one, and I think it's better to say that plainly instead of leaving it ambiguous.
  - Would be nice with a link to the `tf2::Quaternion` class documentation as well, but not sure where it is.
* Feels a bit excessive to inline a [quaternion_multiply](https://github.com/DLu/tf_transformations/blob/main/tf_transformations/__init__.py#L809). Will a first-time reader really want to look at a implementation of quaternion multiplication without any reference?
* About the "_Think in RPY then convert to quaternion_" section. Though there might be some risk of information overload, I think it could be worth to briefly mention that
  - this act of "rotating around each axes" is essentially* applying Euler angles (this term is not mentioned in text)
  - the adopted convention in tf2 (and ROS 2 broadly) is [_fixed axis RPY_](https://github.com/ros2/geometry2/blob/kilted/tf2/include/tf2/LinearMath/Quaternion.hpp#L84-L89) (aka `sxyz`, so rotation around X, then Y, then Z).
  - there exists [other conventions](https://github.com/DLu/tf_transformations/blob/main/tf_transformations/__init__.py#L677) (**s**tatic vs **r**elative, and ordering e.g. XYZ vs ZYZ etc)

I'll think about it some more and make a draft PR into https://github.com/ros2/ros2_documentation. I'd love to hear other peoples thoughts on this though.

danielcranston, 2025-05-11:
PR was merged, I think we can close this as done.

Yadunund, 2025-05-11:
Thanks for testing!

closed 2025-05-11 by Yadunund