ros2/kilted_tutorial_party #374: Recording A Bag From Your Own Node Py

tfoote, 2025-05-01:
Check the documentation for the 'Recording A Bag From Your Own Node Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Recording A Bag From Your Own Node Py page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-03:
i am testing this

Tanishq30052002, 2025-05-03:
- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

Recording
[image]

Playing

[image]

[image]