ros2/kilted_tutorial_party #375: Recording And Playing Back Data

tfoote, 2025-05-01:
Check the documentation for the 'Recording And Playing Back Data' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Recording And Playing Back Data page](https://docs.ros.org/en/kilted/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JatinVira, 2025-05-02:
Happt to take this up,
Please assign to me.

ahcorde, 2025-05-02:
Please take only one issue, when this is done you can choose another one

JatinVira, 2025-05-03:
### Setup
----
```
   PLATFORM INFORMATION
system           : Linux
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

JatinVira, 2025-05-03:
I am working on this as the other assigned ticket is blocked at the moment for your review @ahcorde

JatinVira, 2025-05-03:
[image]

[image]

[image]

- I was able to follow the documentation successfully.
- The documentation seemed clear and concise to me.
- I didn't find any obvious issues/errors in the documentation.

Please also look at the pictures to verify the same.

Please @ahcorde assign this to me and close this

Thankyou.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde