ros2/kilted_tutorial_party #376: RHEL Development Setup

tfoote, 2025-05-01:
Check the documentation for the 'RHEL Development Setup' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RHEL Development Setup page](https://docs.ros.org/en/kilted/Installation/Alternatives/RHEL-Development-Setup.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-04:
I guess this one doesn't make sense, does it ?

danielcranston, 2025-05-13:
Building from source on RHEL (almalinux) using this doc was tested in https://github.com/ros2/kilted_tutorial_party/issues/296#issuecomment-2860575648, so I think this can be closed unless someone else also wants to give it ago.

JatinVira, 2025-05-14:
I tried this out using the ` almalinux:9` docker image as suggested earlier

And hit an error when doing rosdep update and an error further down
Here's the output:

```
[root@71f444b974e4 ros2_kilted]# sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-7.3.0 urdfdom_headers"
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
rosdep detected OS: [almalinux] aliasing it to: [rhel]
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "iron"
Skip end-of-life distro "jade"
Add distro "jazzy"
rosdep detected OS: [almalinux] aliasing it to: [rhel]
Add distro "kilted"
rosdep detected OS: [almalinux] aliasing it to: [rhel]
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Skip end-of-life distro "melodic"
Add distro "noetic"
rosdep detected OS: [almalinux] aliasing it to: [rhel]
Add distro "rolling"
rosdep detected OS: [almalinux] aliasing it to: [rhel]
updated cache in /root/.ros/rosdep/sources.cache
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
rosdep detected OS: [almalinux] aliasing it to: [rhel]
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
rqt_msg: Cannot locate rosdep definition for [rqt_console]
[root@71f444b974e4 ros2_kilted]# cd ~/ros2_kilted/
colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Starting >>> google_benchmark_vendor
Starting >>> osrf_testing_tools_cpp
Starting >>> iceoryx_hoofs
Starting >>> urdfdom_headers
Finished <<< gtest_vendor [0.59s]                                                                       
Starting >>> gmock_vendor
Finished <<< gmock_vendor [0.12s]                                                 
Finished <<< urdfdom_headers [0.98s]                                              
Finished <<< osrf_pycommon [1.08s]                                                
Finished <<< ament_package [1.15s]                                                
Starting >>> ament_cmake_core
Finished <<< ament_lint [1.27s]                                                                       
Starting >>> ament_flake8
--- stderr: iceoryx_hoofs                                                         
In file included from /root/ros2_kilted/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/include/iceoryx_hoofs/internal/posix_wrapper/access_control.hpp:23,
                 from /root/ros2_kilted/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/source/posix_wrapper/access_control.cpp:18:
/root/ros2_kilted/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/platform/linux/include/iceoryx_hoofs/platform/acl.hpp:20:10: fatal error: sys/acl.h: No such file or directory
   20 | #include <sys/acl.h>
      |          ^~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/iceoryx_hoofs.dir/build.make:300: CMakeFiles/iceoryx_hoofs.dir/source/posix_wrapper/access_control.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/iceoryx_hoofs.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
---
Failed   <<< iceoryx_hoofs [2.09s, exited with code 2]
Aborted  <<< osrf_testing_tools_cpp [2.11s]
Aborted  <<< ament_flake8 [0.97s]                                                            
Aborted  <<< ament_cmake_core [1.15s]                                                        
Aborted  <<< google_benchmark_vendor [21.4s]                                     

Summary: 6 packages finished [23.2s]
  1 package failed: iceoryx_hoofs
  4 packages aborted: ament_cmake_core ament_flake8 google_benchmark_vendor osrf_testing_tools_cpp
  2 packages had stderr output: google_benchmark_vendor iceoryx_hoofs
  352 packages not processed
[root@71f444b974e4 ros2_kilted]# 

```

JatinVira, 2025-05-14:
Hey @ahcorde 
 I tried this out on an `almalinux:9` docker image.
And built things from source.

Here are some snippets

`ros2 doctor`
[image]

`architecture and os`
[image]

[image]

`turtlesim`
[image]

`topics`
[image]

Summary:

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

This is good to go.

Aniket is testing out the same on `rockylinux`
Lets wait for him and then we can close this

ahcorde, 2025-05-14:
Thank you for testing

closed 2025-05-14 by ahcorde