ros2/kilted_tutorial_party #377: RHEL Install Binary

tfoote, 2025-05-01:
Check the documentation for the 'RHEL Install Binary' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RHEL Install Binary page](https://docs.ros.org/en/kilted/Installation/Alternatives/RHEL-Install-Binary.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-04:
I guess this one doesn't make sense, does it ?

JatinVira, 2025-05-10:
@ahcorde or @Yadunund
The tag of noble is incorrect .

It should instead be set to rhel

mjcarroll, 2025-05-12:
Description and labels updated.

Aniketkumarroy, 2025-05-13:
Hi I tested this on `almalinux:9` here are the results

[image]
### **Setup**

[image]

[image]

closed 2025-05-14 by fujitatomoya