ros2/kilted_tutorial_party #378: RHEL Install RPMs

tfoote, 2025-05-01:
Check the documentation for the 'RHEL Install RPMs' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RHEL Install RPMs page](https://docs.ros.org/en/kilted/Installation/RHEL-Install-RPMs.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-04:
I guess this one doesn't make sense, does it ?

JatinVira, 2025-05-10:
@ahcorde or @Yadunund 
The tag of `noble` is incorrect .

It should instead be set to `rhel`

JatinVira, 2025-05-10:
I dont have licence to redhat but still gave it a shot in a docker container

```
[root@a58590774d65 /]# sudo dnf update
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

Warning: failed loading '/etc/yum.repos.d/centos-stream-crb.repo', skipping.
Last metadata expiration check: 0:04:41 ago on Sat 10 May 2025 09:58:40 AM UTC.
Dependencies resolved.
Nothing to do.
Complete!
[root@a58590774d65 /]# sudo dnf install ros-kilted-desktop
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

Warning: failed loading '/etc/yum.repos.d/centos-stream-crb.repo', skipping.
Last metadata expiration check: 0:04:52 ago on Sat 10 May 2025 09:58:40 AM UTC.
Error: 
 Problem: package ros-kilted-desktop-0.12.0-2.el9.20250508.230037.x86_64 from ros2-testing requires ros-kilted-turtlesim, but none of the providers can be installed
  - conflicting requests
  - nothing provides qt5-qtbase needed by ros-kilted-turtlesim-1.9.2-2.el9.20250508.221838.x86_64 from ros2-testing
  - nothing provides qt5-qtbase-gui needed by ros-kilted-turtlesim-1.9.2-2.el9.20250508.221838.x86_64 from ros2-testing
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
[root@a58590774d65 /]# sudo dnf install ros-kilted-desktop

```

I think it is conflicting with me not having access to the proper licences.
Maybe someone with a valid license can try this out
but I run into this issue

cottsay, 2025-05-13:
CentOS Stream is "upstream" of RHEL 9, which is the target version for these packages. It would be like trying to use Ubuntu packages on Debian Sid.

An appropriate docker base would be `almalinux:9` or `rockylinux:9`, or a proper RHEL container if you have access to a developer license.

Aniketkumarroy, 2025-05-13:
Hi I tested on `almalinux:9` and here are the results

[image]
### **Setup**

[image]

closed 2025-05-14 by fujitatomoya

JatinVira, 2025-05-14:
Hey, I was also working on this using a `almalinux:9` docker image as suggested by @cottsay 

Here's what I got:

- I was able to follow the documentation
- The documentation seemed clear to me
- The documentation didn't have any obvious errors.

This is already closed, but I was wondering if I could also be a fellow assignee on this issue @ahcorde