ros2/kilted_tutorial_party #379: RMW Implementations

tfoote, 2025-05-01:
Check the documentation for the 'RMW Implementations' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RMW Implementations page](https://docs.ros.org/en/kilted/Installation/RMW-Implementations.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
I read the documentation and everything seems OK, it seems relatively straight forward and I can't spot any errors, but I am not sure if there is anything in particular that I need to test here? The docs point to the DDS/non-DDS RMW implementations but there are already tickets that cover those pages

fujitatomoya, 2025-05-05:
@jncfa thanks for checking on this. there are dedicated issues to check each rmw implementation as you pointed out. i think we can close this issue.

closed 2025-05-05 by fujitatomoya