ros2/kilted_tutorial_party #38: Check ROS 2 doctor CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS

ShahazadAbdulla, 2025-05-03:
Testing Issue #38 on Windows 11 / amd64 / binary / CycloneDDS.

Encountered failures on several checks:

**Check `-r` (`ros2 doctor -r`) - Partial Failure:**
- The command ran but produced warnings about failing to load NetworkReport, PackageReport, PlatformReport, and RosdistroReport due to `No module named 'rosdistro'`.
- Only RMW, Topic, and QoS sections were displayed.

**Check `-rf` (`ros2 doctor -rf`) - Failure:**
- Reported `1/3 check(s) failed`.
- Logged warnings about `No loopback IP address`, `No multicast IP address`, and failure to load PackageCheck/PlatformCheck due to `No module named 'rosdistro'`.
- Failed module reported as: `network`.

**Check `-iw` (`ros2 doctor -iw`) - Failure:**
- Reported `1/3 check(s) failed`.
- Showed the same warnings and failures as `-rf`.

**Check `hello` (`ros2 doctor hello`) - Success:**
- Command ran and printed the expected repeating "MULTIMACHINE COMMUNICATION SUMMARY" blocks successfully. Stopped with Ctrl+C.

**Conclusion:** The `ros2 doctor` command appears to be missing the `rosdistro` Python dependency in the Kilted Windows binary installation, preventing several checks and reports from running. There also seem to be issues with the network checks (`-rf`/`-iw`) detecting loopback/multicast IPs on this Windows setup.

**Full Terminal Output:**
<details>
<summary>Click to expand</summary>
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor -r
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'

QOS COMPATIBILITY LIST
compatibility status : No publisher/subscriber pairs found

RMW MIDDLEWARE
middleware name : rmw_cyclonedds_cpp

TOPIC LIST
topic : none
publisher count : 0
subscriber count : 0

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor -rf
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 87: UserWarning: ERROR: No loopback IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 93: UserWarning: No multicast IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 107: UserWarning: Check entry point PackageCheck fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 107: UserWarning: Check entry point PlatformCheck fails to load: No module named 'rosdistro'

1/3 check(s) failed

Failed modules: network
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor -iw
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 87: UserWarning: ERROR: No loopback IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 93: UserWarning: No multicast IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 107: UserWarning: Check entry point PackageCheck fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api_init_.py: 107: UserWarning: Check entry point PlatformCheck fails to load: No module named 'rosdistro'

1/3 check(s) failed

Failed modules: network

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
Hostname Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
Hostname Msg Count /1.0s
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
Hostname Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
Hostname Msg Count /1.0s
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 11
Subscribed from:
Hostname Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
Received from:
Hostname Msg Count /1.0s
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
Hostname Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
Hostname Msg Count /1.0s
^C[INFO] [1746280552.149310000] [rclcpp]: signal_handler(signum=2)
```
</details>

Yadunund, 2025-05-09:
@ShahazadAbdulla thanks for testing. I just noticed that you're testing on Windows 11 while Windows 10 is still the supported Windows platform for Kilted: https://www.ros.org/reps/rep-2000.html#kilted-kaiju-may-2025-november-2026

I'm not sure if that could the be issue but it might be good if someone on Windows 10 can give this a shot.

clalancette, 2025-05-14:
So this is definitely a bug in the pixi.toml file.  In particular, https://github.com/ros2/ros2/blob/rolling/pixi.toml (and the equivalent kilted, jammy, and humble ones) are missing a dependency on `rosdistro`.  We didn't find this earlier because we skip `ros2doctor` tests on Windows due to unrelated issues.

Unfortunately, conda-forge does not have a "native" package of rosdistro.  But it can install things from pypi, so I'll open a PR to add it as a pypi dependency for now.

There is still a bug in that the NetworkReport class isn't functioning; I'll take a look at that as well.

clalancette, 2025-05-14:
All right, with the two fixes linked this is all fixed for me locally.  Looks like 2 genuine bugs too.  @ShahazadAbdulla thanks for the testing.