ros2/kilted_tutorial_party #381: ROS 1 bridge

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/ros1_bridge)

## Checks
- [ ] **Build ros1_bridge**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the README to compile the bridge
       
  
    
  </details>
- [ ] **Run the birdge with talker and listener**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 1.
       
  
    
  </details>
- [ ] **Run the birdge and exchange images**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 2.
       
  
    
  </details>
- [ ] **Run the birdge and AddTwoInts service**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 3.
       
  
    
  </details>
- [ ] **Run the bridge with only selected topics and services.**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 4.
       
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-06:
Testing this ticket

4-legends, 2025-05-06:
@tfoote @ahcorde 

It looks like it may not possible to install Noetic and Kilted on same Ubuntu version.

Do you guys have any recommendation on how can I set this up?

I tried following steps:
1. Install directly through apt, returns unsupported OS 
2. Find Noetic binaries but was not successful
3. Build Noetic from source, catkin-make fails

- Logs only from the failing section
```
[ 33%] Built target rosconsole_backend_interface
[ 50%] Building CXX object CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::initialize()’:
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:169:23: error: cannot convert ‘ros::console::impl::ROSConsoleStdioAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  169 |   logger->addAppender(new ROSConsoleStdioAppender);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::ROSConsoleStdioAppender*
In file included from /usr/include/log4cxx/spi/loggingevent.h:23,
                 from /usr/include/log4cxx/layout.h:23,
                 from /usr/include/log4cxx/appenderskeleton.h:22,
                 from /home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:85:52: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
   85 |                 void addAppender(const AppenderPtr newAppender) override;
      |                                  ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::print(void*, ros::console::Level, const char*, const char*, const char*, int)’:
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:187:98: error: no matching function for call to ‘log4cxx::spi::LocationInfo::LocationInfo(const char*&, const char*&, int&)’
  187 | cedLog(g_level_lookup[level], str, log4cxx::spi::LocationInfo(file, function, line));
      |                                                                                   ^

In file included from /usr/include/log4cxx/logger.h:24:
/usr/include/log4cxx/spi/location/locationinfo.h:93:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo(const log4cxx::spi::LocationInfo&)’
   93 |                 LocationInfo( const LocationInfo& src );
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:93:17: note:   candidate expects 1 argument, 3 provided
/usr/include/log4cxx/spi/location/locationinfo.h:87:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo()’
   87 |                 LocationInfo();
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:87:17: note:   candidate expects 0 arguments, 3 provided
/usr/include/log4cxx/spi/location/locationinfo.h:79:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo(const char*, const char*, const char*, int)’
   79 |                 LocationInfo( const char* const fileName,
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:79:17: note:   candidate expects 4 arguments, 3 provided
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void* ros::console::impl::getHandle(const std::string&)’:
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:203:36: error: cannot convert ‘log4cxx::LoggerPtr’ {aka ‘std::shared_ptr<log4cxx::Logger>’} to ‘void*’ in return
  203 |   return log4cxx::Logger::getLogger(name);
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
      |                                    |
      |                                    log4cxx::LoggerPtr {aka std::shared_ptr<log4cxx::Logger>}
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘bool ros::console::impl::get_loggers(std::map<std::__cxx11::basic_string<char>, ros::console::levels::Level>&)’:
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:219:120: error: conversion from ‘log4cxx::spi::LoggerRepository*’ to non-scalar type ‘log4cxx::spi::LoggerRepositoryPtr’ {aka ‘std::shared_ptr<log4cxx::spi::LoggerRepository>’} requested
  219 | epo = log4cxx::Logger::getLogger(ROSCONSOLE_ROOT_LOGGER_NAME)->getLoggerRepository();
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::register_appender(ros::console::LogAppender*)’:
/home/legends/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:361:23: error: cannot convert ‘ros::console::impl::Log4cxxAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  361 |   logger->addAppender(g_log4cxx_appender);
      |                       ^~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::Log4cxxAppender*
/usr/include/log4cxx/logger.h:85:52: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
[... cut ...]
/usr/include/log4cxx/logger.h:1614:54: note:   no known conversion for argument 1 from ‘ros::console::impl::Log4cxxAppender*’ to ‘const log4cxx::LogString&’ {aka ‘const std::__cxx11::basic_string<char>&’}
 1614 |                 void removeAppender(const LogString& name) override;
      |                                     ~~~~~~~~~~~~~~~~~^~~~
make[2]: *** [CMakeFiles/rosconsole_log4cxx.dir/build.make:76: CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:368: CMakeFiles/rosconsole_log4cxx.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

4-legends, 2025-05-06:
PR to update readme with compatibility matrix https://github.com/ros2/ros1_bridge/pull/452
@Yadunund

ahcorde, 2025-05-06:
Thank you for testing a openning the PR in the `ros1_bridge` repository.

closed 2025-05-06 by ahcorde