ros2/kilted_tutorial_party #382: ROS 1 bridge

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/ros1_bridge)

## Checks
- [ ] **Build ros1_bridge**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the README to compile the bridge
       
  
    
  </details>
- [ ] **Run the birdge with talker and listener**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 1.
       
  
    
  </details>
- [ ] **Run the birdge and exchange images**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 2.
       
  
    
  </details>
- [ ] **Run the birdge and AddTwoInts service**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 3.
       
  
    
  </details>
- [ ] **Run the bridge with only selected topics and services.**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Follow the instructions in the readme to run Example 4.
       
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-06:
Commenting for anyone looking at this ticket

There is/might be an issue with setting up kilted and noetic to run these test

More information on #381

Yadunund, 2025-05-06:
Closing as this ticket was created due to an error in the requirements config. This has been fixed for future iterations: https://github.com/osrf/ros2_test_cases/pull/1560/commits/61da21aec0292b70020f27a35ca0ec7a4781caa2

closed 2025-05-06 by Yadunund