ros2/kilted_tutorial_party #383: ROS2 Tracing Trace and Analyze

tfoote, 2025-05-01:
Check the documentation for the 'ROS2 Tracing Trace and Analyze' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [ROS2 Tracing Trace and Analyze page](https://docs.ros.org/en/kilted/Tutorials/Advanced/ROS2-Tracing-Trace-and-Analyze.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hey, I found an error here

When I tried to clone using `git clone https://github.com/ros-tracing/tracetools_analysis.git -b kilted` 
I got

`fatal: Remote branch kilted not found in upstream origin`

JatinVira, 2025-05-10:
@suchetanrs is correct.
I tried the tutorial myself and found that the branch of `tracetools_analysis` doesn't yet have a `kilted` specific branch to clone

```
root@573d867530a1:/# source /opt/ros/kilted/setup.bash
ros2 run tracetools status
Tracing enabled
root@573d867530a1:/# 

```

```
root@573d867530a1:~/tracing_ws/src# git clone https://github.com/ros-tracing/tracetools_analysis.git -b kilted
Cloning into 'tracetools_analysis'...
fatal: Remote branch kilted not found in upstream origin
root@573d867530a1:~/tracing_ws/src# 
```

here's the snap from the repo

[image]

@ahcorde 
This is an issue which needs fixing.

JatinVira, 2025-05-12:
@fujitatomoya can you look at this and give me a direction?
Ill work on it

JatinVira, 2025-05-13:
@Yadunund

cottsay, 2025-05-13:
@christophebedard, it looks like the tutorial specifically calls for a `kilted` branch of `tracetools_analysis`. Would you mind creating one (and possibly bumping the `rolling` version if you see fit)?

christophebedard, 2025-05-13:
Thanks for the ping. I've created a `kilted` branch: https://github.com/ros-tracing/tracetools_analysis/tree/kilted.

This repo/package has pretty infrequent changes, so I'll bump it down the road when `rolling` has a new release.

christophebedard, 2025-05-15:
> I proceeded to clone the `rolling` branch just to test and there were a few more errors
> 
> ```
> test_ros2trace_analysis: Cannot locate rosdep definition for [test_tracetools]
> ```

What `rosdep` command did you run?

> ```
> root@cluttered:~/tracing_ws# echo $ROS_DISTRO 
> kilted
> root@cluttered:~/tracing_ws# colcon build --packages-select performance_test --cmake-args -DPERFORMANCE_TEST_RCLCPP_ENABLED=ON
> Starting >>> performance_test
> --- stderr: performance_test                         
> CMake Error at CMakeLists.txt:517 (message):
>   Unsupported ROS_DISTRO
> 
> 
> ---
> Failed   <<< performance_test [3.25s, exited with code 1]
> 
> Summary: 0 packages finished [3.42s]
>   1 package failed: performance_test
>   1 package had stderr output: performance_test
> root@cluttered:~/tracing_ws# 
> ```

`performance_test` just doesn't recognize Kilted: https://gitlab.com/ApexAI/performance_test/-/issues/182.

JatinVira, 2025-05-16:
I tried the same thing and got the exact same error while running `rosdep update` and `colcon build` like @christophebedard

suchetanrs, 2025-05-18:
> What `rosdep` command did you run?

I followed the doc and ran `rosdep update` followed by `rosdep install --from-paths src --ignore-src -y` 
I also tried `rosdep install --from-paths src --ignore-src -y --rosdistro kilted` but it gave me the same error.

christophebedard, 2025-05-19:
Thanks for the info.

I think it's because `test_ros2trace_analysis` is in the workspace and it depends on `test_tracetools`. However, `test_tracetools` isn't in the workspace, since we rely on ROS 2 binaries, and `test_tracetools` is not released, since it's a test-only package: https://github.com/ros2-gbp/ros2_tracing-release/blob/f31660bb027b95d78647da17dd96ece7abe3596e/kilted.ignored#L2.

It's a bit hard for me to test right now, but can one of you try adding `--skip-keys test_tracetools` to the `rosdep install` command? We could just removed the `test_ros2trace_analysis` package from the workspace, but adding this to the `rosdep install` command is a bit simpler.

JatinVira, 2025-05-20:
Hey Thanks for the headsup @christophebedard 
I tried adding the `--skip-keys test_tracetools` to the `rosdep install` command and it worked properly

here's the logs
 
rosdep update
```
root@f522ec73fceb:~/tracing_ws# rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Warning: running 'rosdep update' as root is not recommended.
  You should run 'sudo rosdep fix-permissions' and invoke 'rosdep update' again without sudo.
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "iron"
Skip end-of-life distro "jade"
Add distro "jazzy"
Add distro "kilted"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Skip end-of-life distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
```
rosdep install
```
root@f522ec73fceb:~/tracing_ws# rosdep install --from-paths src --ignore-src -y --skip-keys test_tracetools
executing command [apt-get install -y ros-kilted-ament-mypy]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  python3-mypy python3-mypy-extensions
The following NEW packages will be installed:
  python3-mypy python3-mypy-extensions ros-kilted-ament-mypy
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 8358 kB of archives.
After this operation, 39.0 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu noble/universe amd64 python3-mypy-extensions all 1.0.0-1 [6148 B]
Get:2 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 ros-kilted-ament-mypy amd64 0.19.2-2noble.20250424.200351 [13.8 kB]
Get:3 http://archive.ubuntu.com/ubuntu noble/universe amd64 python3-mypy amd64 1.9.0-4ubuntu1 [8338 kB]
Fetched 8358 kB in 2s (3609 kB/s)       
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python3-mypy-extensions.
(Reading database ... 156075 files and directories currently installed.)
Preparing to unpack .../python3-mypy-extensions_1.0.0-1_all.deb ...
Unpacking python3-mypy-extensions (1.0.0-1) ...
Selecting previously unselected package python3-mypy.
Preparing to unpack .../python3-mypy_1.9.0-4ubuntu1_amd64.deb ...
Unpacking python3-mypy (1.9.0-4ubuntu1) ...
Selecting previously unselected package ros-kilted-ament-mypy.
Preparing to unpack .../ros-kilted-ament-mypy_0.19.2-2noble.20250424.200351_amd64.deb ...
Unpacking ros-kilted-ament-mypy (0.19.2-2noble.20250424.200351) ...
Setting up python3-mypy-extensions (1.0.0-1) ...
Setting up python3-mypy (1.9.0-4ubuntu1) ...
[... 141 lines cut ...]
Setting up python3-urllib3 (2.0.7-1ubuntu0.1) ...
Setting up systemd-hwe-hwdb (255.1.4) ...
Setting up chromium-browser (2:1snap1-0ubuntu2) ...
update-alternatives: using /usr/bin/chromium-browser to provide /usr/bin/x-www-browser (x-www-browser) in auto mode
update-alternatives: using /usr/bin/chromium-browser to provide /usr/bin/gnome-www-browser (gnome-www-browser) in auto mode
Setting up python3-exceptiongroup (1.2.0-1) ...
Setting up python3-wsproto (1.2.0-1) ...
Setting up chromium-chromedriver (2:1snap1-0ubuntu2) ...
Setting up python3-trio (0.24.0-1ubuntu1) ...
Setting up python3-trio-websocket (0.11.1-1) ...
Setting up python3-selenium (4.18.1+dfsg-1) ...
Processing triggers for dbus (1.14.10-4ubuntu4.1) ...
Processing triggers for udev (255.4-1ubuntu8.6) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for libc-bin (2.39-0ubuntu8.4) ...
executing command [apt-get install -y chromium-browser]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
chromium-browser is already the newest version (2:1snap1-0ubuntu2).
chromium-browser set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
#All required rosdeps installed successfully
root@f522ec73fceb:~/tracing_ws# 
```

JatinVira, 2025-05-20:
However the `colcon build` didn't work out

```
root@f522ec73fceb:~/tracing_ws# colcon build --packages-select performance_test --cmake-args -DPERFORMANCE_TEST_RCLCPP_ENABLED=ON
Starting >>> performance_test
--- stderr: performance_test                         
CMake Error at CMakeLists.txt:517 (message):
  Unsupported ROS_DISTRO

---
Failed   <<< performance_test [3.35s, exited with code 1]

Summary: 0 packages finished [3.47s]
  1 package failed: performance_test
  1 package had stderr output: performance_test
root@f522ec73fceb:~/tracing_ws# 
```

christophebedard, 2025-05-20:
> I tried adding the `--skip-keys test_tracetools` to the `rosdep install` command and it worked properly

Thanks for confirming! Would you be willing to open a PR against the https://github.com/ros2/ros2_documentation repo to add `--skip-keys test_tracetools` to the `rosdep install` command?

> However the `colcon build` didn't work out

Can you try it out with this change to the `performance_test` package and let me know if it works? https://gitlab.com/ApexAI/performance_test/-/merge_requests/415

christophebedard, 2025-05-21:
I've opened https://github.com/ros2/ros2_documentation/pull/5651 to add `--skip-keys test_tracetools` to the tutorial.

`performance_test` now supports Kilted (https://gitlab.com/ApexAI/performance_test/-/merge_requests/415 has been merged) and I confirmed that it builds fine.

I think everything has been addressed, so I'll close this issue. Thank you both for testing this!

closed 2025-05-21 by christophebedard