ros2/kilted_tutorial_party #386: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Barry-Xu-2018, 2025-05-08:
PLATFORM INFORMATION
Hardware: Raspberry Pi 4 Model B
System : Ubuntu 24.04.2
processor : aarch64

RMW MIDDLEWARE
middleware name : rmw_connextdds

ROS 2 INFORMATION
distribution name : kilted
distribution type : ros2
distribution status : pre-release
release platforms : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

### Test result 

All tests pass

### Test Steps

Prepare ROS 2 Kilted environment (Refer to https://docs.ros.org/en/kilted/Installation/Ubuntu-Install-Debs.html)

Install the additional required packages
```bash
$ sudo apt install -y rti-connext-dds-7.3.0-ros ros-kilted-rmw-connextdds ros-kilted-demo-nodes-cpp
```

#### Node discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746687713.395619358] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746687714.395572151] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746687715.395567736] [talker]: Publishing: 'Hello World: 3'
...
```

Run `node list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 node list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
/talker
```

Stop demo application and run `node list` again 
```bash
$ ros2 node list
```

#### Topic discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746688525.978130800] [talker]: Publishing: 'Hello World: 1'
[... 66 lines cut ...]

Run `param list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_connextdds
$ ros2 param list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

Stop demo application and run `param list` again 
```bash
$ ros2 param list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
```

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde