ros2/kilted_tutorial_party #387: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

DasariChaitanya, 2025-05-04:
Hi,
I am interested in taking up this ticket and have started the work.

DasariChaitanya, 2025-05-05:
I have performed the tests using two docker containers while using the talker and listener node from demo_nodes_cpp pkg.

### Passed all test cases

> Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.
> 
> ## Setup
> * Rmw: Connext
> * BuildType: Debian
> * Chip: Amd64
> * Os: Noble
>
> ## Checks
  - [x]  **Node discovery**
>   With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

[image]

[image]

- [x]  **Topic discovery**
>   With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

[image]

[image]

- [x]  **Service discovery**
>   With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

[image]

[image]

-  [x]  **Parameter discovery**
>   With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

[image]

[image]

> _You can find the code used to generate this test case [here](https://github.com/audrow/yatm)_

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde