ros2/kilted_tutorial_party #388: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-05:
> Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.
> 
> ## Setup
> * Rmw: Connext
> * BuildType: Debian
> * Chip: Arm64
> * Os: Noble
> 
> ## Links
> * [Repo URL](https://github.com/ros2/ros2cli)
>

SETUP:
- Remote node on Network (**amd64** Ubuntu24.04 Kilted connextdds)

- System Under Test: (arm64 Ubuntu24.04.02LTS Kilted 20250502 connextdds):
```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ source /opt/ros/kilted/setup.bash
ubuntu@kilteddave:~/KiltedDave/ros2ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_connextdds
ubuntu@kilteddave:~/KiltedDave/ros2ws$ uname -a
Linux kilteddave 6.8.0-1024-raspi #28-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr 17 12:20:20 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@kilteddave:~/KiltedDave/ros2ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_LOG_DIR=/home/ubuntu/KiltedDave/ros2ws/roslogs
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted
``` 

 ## Checks
 * [x]  **Node discovery**
>   With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 node list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
/talker                                                                        <<<--- REMOTE NODE
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

and when remote node is stopped:

ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 node list
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

 
* [x]  **Topic discovery**
>   With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 topic list
/chatter                             <<<<---- REMOTE NODE
/parameter_events
/rosout

and when remote node is stopped:

ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 topic list
/parameter_events
/rosout

```

 
* [x]  **Service discovery**
>   With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 service list
/talker/describe_parameters                   <<<<<<<---- REMOTE NODE
/talker/get_parameter_types
/talker/get_parameters
/talker/get_type_description
/talker/list_parameters
/talker/set_parameters
/talker/set_parameters_atomically

and when remote node stopped:

ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 service list
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

 
*  [x]  **Parameter discovery**
>   With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 param list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
/talker:                                                         <<<<--- REMOTE NODE
  start_type_description_service
  use_sim_time
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

and with remote node stopped:

ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 param list
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

* [x] **Agree this ticket may be closed as working correctly.**

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: Connext
BuildType: Debian
OS: Ubuntu Noble
Chip: Arm64

```
>  Node discovery

>> With another node running, use ros2 node list to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

[image]

> Topic discovery

>> With another node running, use ros2 topic list to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

[image]

> Service discovery

>> With another node running, use ros2 service list to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

[image]

> Parameter discovery

>> With another node running, use ros2 param list to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde