ros2/kilted_tutorial_party #389: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Connext
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

zamantousif, 2025-05-14:
I am testing this ticket with source installation on Windows 10 machine with rmw_connextdds.
I ran into the following errors although it seems to work fine with other RMW implementations.

[image]

zamantousif, 2025-05-14:
@ahcorde could you kindly take a look at the above results and let me know if there is something more to try?

zamantousif, 2025-05-20:
Tried this with RTI 7.3.0 as it is the supported version on Windows 10 machine with rmw connext and ros2 source workspace

Terminal 1
```
(pixi_ros2_rolling) C:\dev>ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
C:\dev\kilted\install\Lib\site-packages\launch\logging\__init__.py:149: UserWarning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\zaman\\.ros\\log\\2025-05-21-01-00-26-870054-MSI-4048' -> 'C:\\Users\\zaman\\.ros\\log\\latest'

  warnings.warn(
[INFO] [launch]: All log files can be found below C:\Users\zaman\.ros\log\2025-05-21-01-00-26-870054-MSI-4048
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server.EXE-1]: process started with pid [11588]
[INFO] [robot_state_publisher.EXE-2]: process started with pid [16028]
[INFO] [dummy_joint_states.EXE-3]: process started with pid [22748]
[INFO] [dummy_laser.EXE-4]: process started with pid [3608]
[dummy_laser.EXE-4] [INFO] [1747778428.555446500] [dummy_laser]: angle inc:     0.004363
[dummy_laser.EXE-4] [INFO] [1747778428.555815900] [dummy_laser]: scan size:     1081
[dummy_laser.EXE-4] [INFO] [1747778428.555947000] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher.EXE-2] [INFO] [1747778428.614385600] [robot_state_publisher]: Robot initialized
```
Terminal 2 with the node running
```
(pixi_ros2_rolling) C:\dev>ros2 node list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher

(pixi_ros2_rolling) C:\dev>ros2 service list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
/dummy_joint_states/describe_parameters
/dummy_joint_states/get_parameter_types
/dummy_joint_states/get_parameters
/dummy_joint_states/get_type_description
/dummy_joint_states/list_parameters
/dummy_joint_states/set_parameters
/dummy_joint_states/set_parameters_atomically
/dummy_laser/describe_parameters
/dummy_laser/get_parameter_types
/dummy_laser/get_parameters
/dummy_laser/get_type_description
/dummy_laser/list_parameters
/dummy_laser/set_parameters
/dummy_laser/set_parameters_atomically
/dummy_map_server/describe_parameters
/dummy_map_server/get_parameter_types
/dummy_map_server/get_parameters
/dummy_map_server/get_type_description
/dummy_map_server/list_parameters
/dummy_map_server/set_parameters
/dummy_map_server/set_parameters_atomically
/robot_state_publisher/describe_parameters
/robot_state_publisher/get_parameter_types
/robot_state_publisher/get_parameters
/robot_state_publisher/get_type_description
/robot_state_publisher/list_parameters
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically

(pixi_ros2_rolling) C:\dev>ros2 topic list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
/joint_states
/map
/parameter_events
[... 60 lines cut ...]
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.

(pixi_ros2_rolling) C:\dev>ros2 topic list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
/parameter_events
/rosout

(pixi_ros2_rolling) C:\dev>ros2 service list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.

(pixi_ros2_rolling) C:\dev>ros2 node list
RTI Connext DDS Evaluation License issued to <email> For non-production use only.
Expires on 13-jun-2025
Please contact support@rti.com with any questions or comments.
```
@ahcorde I think this could be closed now