ros2/kilted_tutorial_party #392: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

boby2965, 2025-05-04:
> ## Checks

1st Instance:

terminal :

```
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py

```

2nd Instance:

terminal :

[ x ]  **Node discovery**

```
ubuntu@kilted-desktop:~$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher
ubuntu@kilted-desktop:~$

```

[ x ]  **Topic discovery**

```
ubuntu@kilted-desktop:~$ ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
ubuntu@kilted-desktop:~$ 
```

[ x ]  **Service discovery**

```
ubuntu@kilted-desktop:~$ ros2 service list
/dummy_joint_states/describe_parameters
/dummy_joint_states/get_parameter_types
/dummy_joint_states/get_parameters
/dummy_joint_states/get_type_description
/dummy_joint_states/list_parameters
/dummy_joint_states/set_parameters
/dummy_joint_states/set_parameters_atomically
/dummy_laser/describe_parameters
/dummy_laser/get_parameter_types
/dummy_laser/get_parameters
/dummy_laser/get_type_description
/dummy_laser/list_parameters
/dummy_laser/set_parameters
/dummy_laser/set_parameters_atomically
/dummy_map_server/describe_parameters
/dummy_map_server/get_parameter_types
/dummy_map_server/get_parameters
/dummy_map_server/get_type_description
/dummy_map_server/list_parameters
/dummy_map_server/set_parameters
/dummy_map_server/set_parameters_atomically
/robot_state_publisher/describe_parameters
/robot_state_publisher/get_parameter_types
/robot_state_publisher/get_parameters
/robot_state_publisher/get_type_description
/robot_state_publisher/list_parameters
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically
ubuntu@kilted-desktop:~$
```

[ x ]  **Parameter discovery**

```
ubuntu@kilted-desktop:~$ ros2 param list
/dummy_joint_states:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_laser:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_map_server:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/robot_state_publisher:
  frame_prefix
  ignore_timestamp
  publish_frequency
  qos_overrides./joint_states.subscription.depth
  qos_overrides./joint_states.subscription.history
  qos_overrides./joint_states.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  robot_description
  start_type_description_service
  use_sim_time

```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde