ros2/kilted_tutorial_party #394: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Cyclone
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-01:
PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-24-generic-aarch64-with-glibc2.39
release          : 6.11.0-24-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

**RESULT**

I have ran 2 instance of same installation in a network and tested the checks.
This test have **passed**.

christophebedard, 2025-05-01:
Thanks for testing. Can you include the commands you ran and their output?

pradyum, 2025-05-02:
**1st Instance:**

.bashrc modification:

```
source /opt/ros/kilted/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

terminal: 

```
~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

**2nd Instance:**

.bashrc modification:

```
source /opt/ros/kilted/setup.bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

terminal: 

- [x] Node discovery

```
~$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher
```

- [x] Topic discovery

```
~$ ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
```

- [x] Service discovery

```
~$ ros2 service list
/dummy_joint_states/describe_parameters
/dummy_joint_states/get_parameter_types
/dummy_joint_states/get_parameters
/dummy_joint_states/get_type_description
/dummy_joint_states/list_parameters
/dummy_joint_states/set_parameters
/dummy_joint_states/set_parameters_atomically
/dummy_laser/describe_parameters
/dummy_laser/get_parameter_types
/dummy_laser/get_parameters
/dummy_laser/get_type_description
/dummy_laser/list_parameters
/dummy_laser/set_parameters
/dummy_laser/set_parameters_atomically
/dummy_map_server/describe_parameters
/dummy_map_server/get_parameter_types
/dummy_map_server/get_parameters
/dummy_map_server/get_type_description
/dummy_map_server/list_parameters
/dummy_map_server/set_parameters
/dummy_map_server/set_parameters_atomically
/robot_state_publisher/describe_parameters
/robot_state_publisher/get_parameter_types
/robot_state_publisher/get_parameters
/robot_state_publisher/get_type_description
/robot_state_publisher/list_parameters
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically
```

- [x] Parameter discovery

```
~$ ros2 param list
/dummy_joint_states:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_laser:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_map_server:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/robot_state_publisher:
  frame_prefix
  ignore_timestamp
  publish_frequency
  qos_overrides./joint_states.subscription.depth
  qos_overrides./joint_states.subscription.history
  qos_overrides./joint_states.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  robot_description
  start_type_description_service
  use_sim_time
```

**After turning off the launch nodes of 1st instance the service, node and param list are empty in 2nd instance. The ```ros2 topic list``` is**

```
~$ ros2 topic list 
/parameter_events
/rosout
```

christophebedard, 2025-05-02:
Awesome, thank you!

closed 2025-05-02 by christophebedard