ros2/kilted_tutorial_party #396: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

hyfan1116, 2025-05-06:
### Checks
- [ ] Node discovery
- [ ] Topic discovery
- [x] Service discovery
- [x] Parameter discovery

### Issue
`ros2 node list` and `ros2 topic list` still showing killed node and killed node's topics, even after the node on the other host has been killed.

w/ node running on host 2:
[image]

w/o node running on host 2:
[image]

### Environment
Host 1: Kilted ROS2 container on Ubuntu 24, FastDDS
Host 2: Kilted ROS2 container on Ubuntu 20, FastDDS

Aniketkumarroy, 2025-05-11:
Hi I  tested this
### While running a node
Machine 1

[image]
Machine 2

[image]
### While stopping the node
Macine 1

[image]

Machine 2

[image]
### System setup
Machine 1

[image]
Machine 2

[image]
### Summary
- [x] Node discovery
- [x] Topic discovery
- [x] Service discovery
- [x] Parameter discovery

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde