ros2/kilted_tutorial_party #397: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

GeorgeVrk, 2025-05-05:
Hello,

- [x] Node discovery

Windows :
```
C:\ros2cli>ros2 node list
/turtlesim
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 run turtlesim turtlesim_node 
QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1746465707.434086812] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746465707.453263694] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]

```

[image]

Windows :
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run turtlesim turtlesim_node
[INFO] [1746465830.571689000] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746465830.579451600] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 node list
/turtlesim
```

[image]

- [x] Topic Discovery

Windows :
```
C:\ros2cli>ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 run turtlesim turtlesim_node 
QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1746466901.673682103] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466901.692011617] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]

```

[image]

Windows :
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run turtlesim turtlesim_node
[INFO] [1746466152.362134100] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466152.370195900] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose

```

[image]

- [x] Service discovery

Windows :
```
C:\ros2cli>ros2 service list
/clear
/kill
/reset
/spawn
/turtle1/set_pen
/turtle1/teleport_absolute
/turtle1/teleport_relative
/turtlesim/describe_parameters
/turtlesim/get_parameter_types
/turtlesim/get_parameters
/turtlesim/get_type_description
/turtlesim/list_parameters
/turtlesim/set_parameters
/turtlesim/set_parameters_atomically
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 run turtlesim turtlesim_node 
QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1746466248.711299618] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466248.729608460] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

[image]

Windows :
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run turtlesim turtlesim_node
[INFO] [1746466340.883533500] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466340.891480200] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 service list
/clear
/kill
/reset
/spawn
/turtle1/set_pen
/turtle1/teleport_absolute
/turtle1/teleport_relative
/turtlesim/describe_parameters
/turtlesim/get_parameter_types
/turtlesim/get_parameters
/turtlesim/get_type_description
/turtlesim/list_parameters
/turtlesim/set_parameters
/turtlesim/set_parameters_atomically
```

[image]

- [x] Parameter discovery

Windows :
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 run turtlesim turtlesim_node 
QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1746466625.634919265] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466625.652122626] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

[image]

Windows :
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 run turtlesim turtlesim_node
[INFO] [1746466781.886311900] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1746466781.894074400] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
```

Noble :
```
georgev@goofy-SATELLITE-PRO-R50-E:~$ ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time

```

[image]

Sopping the nodes no longer shows the nodes/topics/services/parameters.
I've also tested these without RDP just to be sure and the output is the same.

Yadunund, 2025-05-05:
Thanks for your contribution!

closed 2025-05-05 by Yadunund

knmcguire, 2025-05-09:
@GeorgeVrk Just a question here, what do you use to go around the firewall issues on windows? I'm trying some of the networking tickets and trying to recreate this but windows is not seeing any of the nodes on my rp5 with the kilted docker

GeorgeVrk, 2025-05-09:
Hi @knmcguire,

I tried to avoid the complexity of configuring the firewall rules since I'm still new with ROS2 and networking, so I temporarily disabled the windows firewall to allow node discovery. This was purely to confirm the setup was working.

I could revisit, allowing specific UDP ports through the firewall and let you know if you'd like :).

knmcguire, 2025-05-09:
Ah no worries. I've already gone through the UDP port routes before and multiple ports are used for discovery which makes it even more complex. Someone advised using a VPN network so I'll try that. Thanks!