ros2/kilted_tutorial_party #4: Adding Physical and Collision Properties to a URDF Model

tfoote, 2025-05-01:
Check the documentation for the 'Adding Physical and Collision Properties to a URDF Model' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Adding Physical and Collision Properties to a URDF Model page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Adding-Physical-and-Collision-Properties-to-a-URDF-Model.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-03:
Testing this ticket

4-legends, 2025-05-03:
- Setup
```
legends@4legends-ASUS:~/ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

legends@4legends-ASUS:~/ros2_ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp

``` 

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.
- [x] I ran the RVIZ2 with 07-physics.urdf and it worked without any issues.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde