ros2/kilted_tutorial_party #400: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2025-05-01:
@pradyum thanks for taking care of this, can you tap the all checkbox in the issue header?

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: Fastdds
BuildType: Debian
OS: Ubuntu Noble
Chip: Arm64

```
>  Node discovery

>> With another node running, use ros2 node list to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

[image]

> Topic discovery

>> With another node running, use ros2 topic list to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

[image]

> Service discovery

>> With another node running, use ros2 service list to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

[image]

> Parameter discovery

>> With another node running, use ros2 param list to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde