ros2/kilted_tutorial_party #401: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-08:
I am testing it out

Tanishq30052002, 2025-05-08:
can i test this functionality in this docker image?

https://hub.docker.com/r/redhat/ubi9

Aniketkumarroy, 2025-05-13:
Hi I used `almalinux:9` and here are the results

[image]
### **Setup**

[image]

[image]

fujitatomoya, 2025-05-14:
@Aniketkumarroy thanks for testing and results.