ros2/kilted_tutorial_party #403: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

akky20, 2025-05-06:
Interested

akky20, 2025-05-06:
Done with the testing

## Setups 
Rmw: Zenoh
Chip: Amd64
Os: Ubuntu 24. Noble
[image]

## Node  

[image]

## Topic 

[image]

## Service 

[image]

## parameter 

[image]

ahcorde, 2025-05-06:
Thank you for testing

closed 2025-05-06 by ahcorde