ros2/kilted_tutorial_party #404: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

knmcguire, 2025-05-09:
Hi! Currently testing this out with a windows 11 and RP5 with ubuntu 24.04. 

None of the examples are working even with firewall turned off.... Currently debugging the issue here and have tried [these [instructions](https://github.com/ros2/rmw_zenoh?tab=readme-ov-file#connecting-multiple-hosts) as well

knmcguire, 2025-05-12:
Credit to @JEnoch for helping me to get this to work. Over network, the zenoh routers need to connect to eachother on different hosts but I had difficulty of having my RP5 zenoh router to connect to the windows zenoh router. But other way worked!

On RP5 with noble+ kilted (host 1)

RP5 - Terminal 1  
```
ros2 run rmw_zenoh_cpp rmw_zenohd
```

Win11 - Terminal 1 Connect to zenoh router on RP5 (ip 192.168.0.1)
```
set ZENOH_CONFIG_OVERRIDE=connect/endpoints=["tcp/192.168.0.1:7447"]  
ros2 run rmw_zenoh_cpp rmw_zenohd
```

RP5 - terminal 2: 
```
RMW_IMPLEMENTATION=rmw_zenoh_cpp ros2 run turtlesim turtlesim_node
```

Win11 - Terminal 2

```
(pixi_ros2_rolling) C:\pixi_ws>set RMW_IMPLEMENTATION=rmw_zenoh_cpp

(pixi_ros2_rolling) C:\pixi_ws>ros2 node list
/turtlesim

(pixi_ros2_rolling) C:\pixi_ws>ros2 topic list
/parameter_events
/rosout
/turtle1/cmd_vel
/turtle1/color_sensor
/turtle1/pose

(pixi_ros2_rolling) C:\pixi_ws>ros2 service list
/clear
/kill
/reset
/spawn
/turtle1/set_pen
/turtle1/teleport_absolute
/turtle1/teleport_relative
/turtlesim/describe_parameters
/turtlesim/get_parameter_types
/turtlesim/get_parameters
/turtlesim/get_type_description
/turtlesim/list_parameters
/turtlesim/set_parameters
/turtlesim/set_parameters_atomically

(pixi_ros2_rolling) C:\pixi_ws>ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

So it works with the windows zenoh router connecting explicitly to the external host zenoh router. It however doesn't work to initialize the connection the other way around but perhaps outside of the scope of this test.